RobotLocomotion/drake #8997: Enable Bazel remote caching on more/most continuous/experimental builds by default

jamiesnape, 2018-06-14:
A tracking issue for the above. More relevant when bionic support causes the number of jobs to increase more than our CI budget can handle.

FYI @EricCousineau-TRI.

EricCousineau-TRI, 2018-12-03:
Per f2f, can close when all PR jobs can use caching (including debug / sanitizer).

At present, we save some budget, but dev experience is still a little painful due to having to wait for presently non-cached builds (debug, sanitizer).

jwnimmer-tri, 2019-02-15:
FYI At TRI I'm experimenting with turning on Fission for Anzu.  Fission should work fine on Bionic at least, and it's on the order of 5x reduction in debug build footprints (80% smaller).  I think upstream https://github.com/bazelbuild/bazel/pull/6606 would enable Fission by default on all of our platforms, but it's stalled in review.  I wonder if we could help nudge that along, or if there is a 30-line patch to Drake that could have the same effect (declare our own crosstool, but delegate most of it to upstream local_cc).

Fission may also help with #9952.

jamiesnape, 2019-02-15:
(My understanding is that debug fission would not work with the Apple compiler at present, but the compiler test in the Bazel PR may erroneously indicate that it is.)

jamiesnape, 2019-02-15:
https://github.com/bazelbuild/bazel/issues/327 is relevant for Mac.

jwnimmer-tri, 2019-02-15:
Fair enough.  Sounds like the low-hanging fruit might be a bionic debug build that is included in the pre-merge set, depending on the crosstool hijinks that we'd need.  How about I take a bit more time to see how Anzu fares once I turn this on there, and then we can revisit the question for Drake?

jamiesnape, 2019-02-15:
Sounds good.

jwnimmer-tri, 2020-05-14:
> Per f2f, can close when all PR jobs can use caching (including debug / sanitizer) ...

I just looked at the most recent PR, and all of the jobs appear to be using the remote cache.

closed 2020-05-14 by jwnimmer-tri