RobotLocomotion/drake #9011: Question about the file name atlas_command_t.hpp

junhyeokahn, 2018-06-15:
Hello,
I am looking at the codes for humanoid controller(valkyrie) and I saw several files includes ```#include "lcmtypes/bot_core/atlas_command_t.hpp"```.
But I cannot find any file called ```atlas_command_t.hpp```.
I guess I am asking this because of the lack of knowledge about lcm but could you briefly explain where it is and what does it do?

sammy-tri, 2018-06-18:
That file is autogenerated from the bot_core_lcmtypes repository (see here for the original input: https://github.com/openhumanoids/bot_core_lcmtypes/blob/master/lcmtypes/bot_core_atlas_command_t.lcm ).  For general questions about drake (as opposed to bugs or feature requests), it's usually preferred to [post a question on stack overflow with the drake tag](https://stackoverflow.com/questions/ask?tags=drake) (see http://drake.mit.edu/getting_help.html )

junhyeokahn, 2018-06-18:
Thanks for the information :)

closed 2018-06-18 by junhyeokahn