RobotLocomotion/drake #9013: Migrate time-varying LQR from matlab sha

joycezyu, 2018-06-16:
## Description

> Per slack discussion under #dynamics with @hongkai-dai, @RussTedrake, @sherm1, I would like to request the implementation of the time-varying Linear Quadratic Regulator functionality in the C++ version of drake. tvlqr is available in the matlab version of drake, so transferring to c++ shouldn't be too bad.

RussTedrake, 2018-06-18:
n.b. original version is here:
https://github.com/RobotLocomotion/drake/blob/last_sha_with_original_matlab/drake/matlab/systems/%40DrakeSystem/tvlqr.m

i can try to do this sometime in the next few days.

junhyeokahn, 2018-06-22:
It would be also nice to see ```runTrajectorySwingUpAndBalance.m``` in the previous tag in the c++ version of drake.

mpetersen94, 2018-07-18:
Any updates on this?

joycezyu, 2018-07-18:
We have a hacky-ish version of `tvlqr` implemented locally. Hope to PR in the near future!

joycezyu, 2018-08-02:
@junhyeokahn The local version of `tvlqr` lives in TRI for now due to the legal issues... I will post here if things change

akshitj1, 2019-10-21:
Have implemented a draft version for TimeVaryingLQR tested with perching glider [here](https://github.com/akshitj1/drake_plants/blob/master/src/glider/tvlqr.h). Will raise a PR in near future. Linking as it might be helpful for someone to refer meanwhile.

PR #13078 (2020-04-20): Adds finite horizon linear quadratic regulator
First version -- continuous-time only, Q and R are time-invariant.
Resolves #9013.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13078)
commit 283864f2 (2020-04-20): Adds finite horizon linear quadratic regulator (#13078)

First version -- continuous-time only, Q and R are time-invariant.
Resolves #9013.
closed 2020-04-20 by jwnimmer-tri