RobotLocomotion/drake #9017: Broken link to simple_continuous_time_system.cc in doc/index.rst

hazelnusse, 2018-06-18:
Text "Easily add your own models/components" links to:

https://github.com/RobotLocomotion/drake/blob/master/drake/examples/simple_continuous_time_system.cc

but it should be:
https://github.com/RobotLocomotion/drake/blob/master/examples/simple_continuous_time_system.cc

Patch coming.

hazelnusse, 2018-06-18:
https://github.com/RobotLocomotion/drake/pull/9018

hazelnusse, 2018-06-18:
Hmmm, I guess I didn't need to create an issue then open a pull request.

sherm1, 2018-06-18:
Not necessary when the PR is imminent, but nice to have. Then in the initial PR comment you should use a GitHub magic phrase like "Fixes #9017" ("Resolves" or "Closes" works too) so that the associated issue is automatically closed when the PR is merged.

Thanks!

closed 2018-06-18 by sherm1

hazelnusse, 2018-06-18:
Got it, thanks.