RobotLocomotion/drake #9039: `monolithic_pick_and_place_system_test` times out in nightly

ggould-tri, 2018-06-21:
## Description

`monolithic_pick_and_place:monolithic_pick_and_place_system_test` sometimes times out in nightly CI.

Examples at:
 * https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-xenial-gcc-bazel-nightly-everything-release/501
 * https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-xenial-clang-bazel-nightly-everything-release/500/

It is likely that the timeout of this test needs to be incremented.  It is possible that the performance of this test has regressed.

ggould-tri, 2018-06-22:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-bazel-continuous-everything-release/2396/console

ggould-tri, 2018-06-22:
Again: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-bazel-continuous-everything-release/2397/console

ggould-tri, 2018-06-25:
Here is a link to a relevant discussion of the test and its failure modes, from slack:

https://drakedevelopers.slack.com/archives/C270MN28G/p1529698496000309

ggould-tri, 2018-06-25:
Three more failures over the weekend:

 * https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-bazel-continuous-everything-release/2398/console
 * https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-bazel-continuous-everything-release/2399/console
 * https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-xenial-gcc-bazel-continuous-everything-release/2400/console

jwnimmer-tri, 2018-06-25:
This test is in a `dev` folder.  It's time to disable it and assign an issue to the author.  I will open a PR.

jwnimmer-tri, 2018-06-25:
For the record, it seems to be mostly (or always?) shard 31 that times out -- the highest-number shard.  I don't know if that is a reporting inaccuracy or if shard 31 is really the slowest one.

jwnimmer-tri, 2019-03-04:
Fixed by #10591 (removing all of pick_and_place).

closed 2019-03-04 by jwnimmer-tri