RobotLocomotion/drake #9040: High Sierra build steps sometimes take unreasonable time

ggould-tri, 2018-06-21:
## Description

The following examples show cases where build steps that normally take less than a second time out at 350 seconds:

 * https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-highsierra-clang-bazel-nightly-debug/230/console : primitives_test and rendering_test each normally take 1 second; they timed out at 366s.
 * https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-highsierra-clang-bazel-nightly-debug/234/console : drakelint steps that normally take 0.4 seconds timed out at 387s.
 * https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-highsierra-clang-bazel-continuous-release/814/console : bazel_codestyle that normally takes a bit less than a second timed out at 350s.
 * https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-highsierra-clang-bazel-continuous-release/830/console : Three buildifier steps that normally takes a bit less than a second timed out at 380s.

I think it is notable that most of these fail on linters.  It seems to me that among our timeout-constrained build steps, linters probably touch the largest number of files.  So this may be an issue with occasional huge IO latency which by the nature of the statistics will hit linters disproportionately.

ggould-tri, 2018-06-22:
Again: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-highsierra-unprovisioned-clang-bazel-nightly-release/245/console

ggould-tri, 2018-06-25:
Another: https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-highsierra-clang-bazel-continuous-release/840/console

jamiesnape, 2018-06-25:
An early hypothesis was that this was a Mac analog of #8651, though Mac builds are much less parallel, so in reality that is probably less likely.

EricCousineau-TRI, 2019-04-25:
Per f2f, a possibly clean workaround for excessive lint timeouts:
Disable linting on most Mac builds except for one (just to ensure parity with Linux lint checks).

Maybe worth it if it's a 5min solution, which saves ~2min every time a Build Cop checks for lint flakes.

jwnimmer-tri, 2020-03-25:
Another idea: we could easily mark the lint tests as `flaky = True` iff on macOS.  (Possibly only in CI, too.)

EricCousineau-TRI, 2020-03-25:
SGTM. Should we also consider shortening the timeout in that case?

jwnimmer-tri, 2020-03-25:
They are all `size = "small"` already AFAICT.

jwnimmer-tri, 2020-07-30:
Hmm, looks like we don't check the return code from `differ = subprocess.Popen(` in our linter wrapper.  Possibly we could do that and add an in-band retry to deflake within the wrapper itself, instead of asking bazel to do it.  We still do want to fail immediate for true positive lint errors without repeating the linting three times (especially during local development).

https://github.com/RobotLocomotion/drake/blob/413b56b32d01bb79e656d32d57981bf825104674/tools/lint/clang_format_lint.py#L22-L35

SeanCurtis-TRI, 2023-02-28:
Another instance last night in linting (an arbitrary directory):

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-monterey-clang-bazel-nightly-everything-release/143/

svenevs, 2024-01-09:
Possible instance in https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-everything-release/241/consoleFull

```
[4:14:20 AM]  //common:py/_text_logging_no_spdlog_test_private_headers_cc_impl_drakelint FAILED in 0.0s
[4:14:20 AM]    /Users/admin/workspace/mac-arm-ventura-clang-bazel-nightly-everything-release/_bazel_admin/27c18ea9175ffb1de77a562928d9b5a6/execroot/drake/bazel-out/darwin_arm64-opt/testlogs/common/py/_text_logging_no_spdlog_test_private_headers_cc_impl_drakelint/test.log
[4:14:20 AM]  //systems/framework:py/diagram_context_test_drakelint                    FAILED in 0.0s
[4:14:20 AM]    /Users/admin/workspace/mac-arm-ventura-clang-bazel-nightly-everything-release/_bazel_admin/27c18ea9175ffb1de77a562928d9b5a6/execroot/drake/bazel-out/darwin_arm64-opt/testlogs/systems/framework/py/diagram_context_test_drakelint/test.log
```

Linting appeared to fail for reasons unrelated to linting (flaky tests in that execution), no output from linting runs.

liangfok, 2024-05-07:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-debug/108/consoleFull

```
[2:36:49 AM]  ERROR: /Users/admin/workspace/mac-arm-ventura-clang-bazel-nightly-debug/src/geometry/render_gl/BUILD.bazel:353:15: Testing //geometry/render_gl:py/extra_srcs_cpplint_clang_format_lint failed: (Killed): generate-xml.sh failed: error executing TestRunner command (from target //geometry/render_gl:py/extra_srcs_cpplint_clang_format_lint) external/bazel_tools/tools/test/generate-xml.sh bazel-out/darwin_arm64-dbg/testlogs/geometry/render_gl/py/extra_srcs_cpplint_clang_format_lint/test.log ... (remaining 3 arguments skipped)
```

closed 2024-11-01 by jwnimmer-tri