RobotLocomotion/drake #9060: `hermitian_continuous_extension` tests fail on mac

ggould-tri, 2018-06-26:
Tests introduced in #8950 failed on mac CI.  A representative failure example is:

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-highsierra-clang-bazel-continuous-everything-release/795/

hidmic, 2018-06-26:
I think this might be related to a similar issue I faced some time ago (https://github.com/RobotLocomotion/drake/issues/8316). Let me open a PR with the relevant lines shortened.

jwnimmer-tri, 2018-06-26:
Ah, indeed!  We submitted a fix to cpplint (google/styleguide#342) for this, and its been accepted.  However, RobotLocomotion/styleguide#10 is stalled, which means that we haven't merged the newer cpplint into Drake yet, so we are still prone to this bug.

jamiesnape, 2018-06-26:
> Ah, indeed! We submitted a fix to cpplint (google/styleguide#342) for this, and its been accepted. However, RobotLocomotion/styleguide#10 is stalled, which means that we haven't merged the newer cpplint into Drake yet, so we are still prone to this bug.

Do we know what is delaying that? It has been more than two months now.

jwnimmer-tri, 2018-06-26:
That seems like a discussion for RobotLocomotion/styleguide#10's thread.

jamiesnape, 2018-06-26:
True, my apologies.

PR #9062 (2018-06-26): Shortened //systems/analysis/hermitian_continuous_extension.h docs line length
Fixes #9060. Documentation lines are shortened to cope with cpplint bad line counts in Mac, whenever these include Unicode characters (refer to https://github.com/google/styleguide/pull/342 for further details).

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9062)
closed 2018-06-26 by sammy-tri