RobotLocomotion/drake #9064: Taylor expansion for a given expression

joycezyu, 2018-06-26:
Can we have the taylor expansion module built for some given expressions of nonlinear equations? Up to 3rd order would be great. Thanks!

hongkai-dai, 2018-06-26:
To be more specific, the expression to be expanded should be a **symbolic** expression, this is going to be used in funnel/region-of-attraction computation, where we need to approximate the system dynamics with polynomials.

soonho-tri, 2018-06-26:
I think its signature should be something like the following. Can any of you confirm? (or suggest a better one)

```c++
/// Returns the series expansion of `e` around `x0` of order `n`. 
Expression Series(const Expression& e, const Environment& x0, int n)
```

joycezyu, 2018-06-26:
That looks pretty fine by me.

hongkai-dai, 2018-06-26:
Thanks Soonho! That function signature looks great.

Do you think it makes sense to have `Taylor expansion` appear somewhere in either the function name or the documentation? `Series` is not very specific to me.

Also it would be good if we can have a function that we can pass in an Eigen::Vector of expressions, and also return the Eigen::Vector of expressions. This could be easily done by using `unaryExpr`, and would be a nice syntax sugar.

soonho-tri, 2018-06-26:
@hongkai-dai , just updated the signature above.

> Also it would be good if we can have a function that we can pass in an Eigen::Vector of expressions, and also return the Eigen::Vector of expressions. This could be easily done by using unaryExpr, and would be a nice syntax sugar.

I agree. After finishing the scalar version, I'll add the vector version.

joycezyu, 2018-06-26:
There is a function named `DoFirstOrderTaylorApproximation` in `systems/primitives/linear_system` that construct a pointer of type `AffineSystem`. It does quite different things than the `Taylor expansion` intended here - might be something worth considering when deciding on the function name.

joycezyu, 2018-06-28:
@soonho-tri To follow up an earlier discussion, it would be highly desirable to go to the 3rd order. The reason is that we have to guarantee an odd order of the dynamics (`f(x)`) in order to multiply an odd order of `Vdot`. As a result, the product is of even order and it satisfies the definition of Sum-of-Squares.

soonho-tri, 2018-06-29:
@joycezyu , I've implemented it in my branch, https://github.com/soonho-tri/drake/tree/taylor-expand. Can you pick it up and test it with your application?

PR #9082 (2018-07-03): Add TaylorExpand
Close #9064

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9082)
closed 2018-07-03 by sammy-tri