RobotLocomotion/drake #9077: Vector input port type-mismatch error message needs improving

sherm1, 2018-06-29:
@mposa defined a BasicVector input port, plugged it into a MySpecialVector output port (where MySpecialVector derives from BasicVector) and got the following error message when evaluating the input:
```
System::EvalInputValue(): expected value of type BasicVector for 
input port[0] but the actual type was MySpecialVector.
```
That was not very helpful in diagnosing the actual problem which was that 
`EvalInputValue<BasicVector>()` treats the port as abstract-valued and wants an exact type match. A vector port's value should be obtained instead with `EvalVectorInput()` which would have worked.

If we know the port is vector valued and get a type mismatch in EvalInputValue(), the error message should suggest using EvalVectorInput() instead.

([Thread in slack](https://drakedevelopers.slack.com/messages/C2CHRT98E/).)

jwnimmer-tri, 2019-07-02:
I wonder if the revised port-eval APIs have resolved this sufficiently?

sherm1, 2019-07-04:
I'm not sure whether this is fixed. IMO it should stay open until we get a chance to try it.

jwnimmer-tri, 2024-11-01:
Probably fixed, but if not then it's certainly "not planned" since subclassing BasicVector is strongly dis-preferred.

closed 2024-11-01 by jwnimmer-tri