RobotLocomotion/drake #9087: Stop creating $HOME/.sdformat/sdformat.log

jwnimmer-tri, 2018-07-02:
Drake's use of libsdformat means that when we load models, there is a fair chance that `$HOME/.sdformat/sdformat.log` is going to be created (including the directory) and that log file (over-)written.  This is at best a race condition between multiple processes parsing at once, and really logs due to Drake methods should be going through `drake::log()` infrastructure, not blatting out to files.  We should amend libsdformat so that using-code can redirect its Console.

amcastro-tri, 2018-07-02:
cc'ing @nkoenig

jwnimmer-tri, 2018-10-28:
Still broken, but not worth tracking with an issue.

closed 2018-10-28 by jwnimmer-tri

jwnimmer-tri, 2020-08-10:
This bothered me enough in our build logs this morning that I've submitted a hacky fix, and upstream request for non-hacky API.