RobotLocomotion/drake #9092: Upgrade SNOPT binding

hongkai-dai, 2018-07-03:
> Please fill out all quoted sections (`> `) that need a response; feel free
to remove the quoted sections that do not need a response.

> Before you post:

> * Have you reviewed the documentation on
[Getting Help](http://drake.mit.edu/getting_help.html)?

Yes
> * Have you reviewed existing GitHub issues and StackOverflow posts?

Yes
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

Yes

## Description

> Please provide a short description: expected behavior, and what actually
happens. If this is an enhancement request, please provide a concrete use case.

The goal is to be able to run multiple SNOPT calls in parallel. There are several steps we need/can to do.
1. Upgrade SNOPT to newer versions.
   I believe we are using SNOPT 7.2.12 https://github.com/RobotLocomotion/snopt/blob/master/README.WhatsNew. We should upgrade SNOPT to newer versions, like 7.6.
2. We currently use SNOPT C interface, together with f2c (fortran to C). Should we consider to use SNOPT c++ interface? Will the C++ interface support / be better suited with running multiple SNOPT?

@jwnimmer-tri @avalenzu @RussTedrake

hongkai-dai, 2018-07-03:
@RussTedrake , yes. As it turns out we are already on SNOPT 7.6.1 (not the master branch of RobotLocomotion/snopt repo where I looked, but on a separate branch). So we only need to do 2 here.

hongkai-dai, 2018-07-04:
I tried to change all the snopt calls in `snopt_solver.cc` with the API in `snopt-interface`, but I couldn't get the build system to work. As @jwnimmer-tri and I discussed over the email, we will need to modify the build file in order to use snopt-interface. We will do this in two steps:

1. Jeremy or Jamie will use gfortran compiler to compile snopt.
2. I will port `snopt_solver` on top of the step 1.

jwnimmer-tri, 2018-07-06:
What part of this issue it not just a duplicate of #7984?

hongkai-dai, 2018-07-06:
Close the issue as it is a duplicate of #7984.

closed 2018-07-06 by hongkai-dai