RobotLocomotion/drake #9095: Simulator calls to discrete/continuous systems's APIs

amcastro-tri, 2018-07-04:
Our `systems::` simulator seems to be calling `DoMapQDotToVelocity()` (and the reverse) on discrete systems when they are part of a larger `Diagram` composed of both continuous and discrete systems. This causes problems down the line for systems that have the option of being continuous and discrete (like RBT and MBP). The quick solution is that those systems implement a no-ops for the corresponding methods depending on whether they are being modeled as continuous or discrete systems.
This transfers the burden to the author of those systems who now has to check on each method that interacts with the simulator whether the system is discrete or continuous. 
Probably best would be if the simulator would not call `MapQDotToVelocity`/`CalcTimeDerivatives` on discrete systems and would not call `CalcDiscreteVariableUpdates()` on continuous systems?

sherm1, 2018-07-04:
Many systems have a mix of continuous and discrete variables, so the Simulator will generally have to invoke these methods at the top Diagram level. But in the subsequent recursion it would be good if the calls were suppressed for any subsystem (or sub-diagram) that has no variables of the right type.

edrumwri, 2018-07-06:
It's unclear from the description how I can reproduce this problem, and a trace through the system framework code appears to show all the right things being done (i.e., the mapping operations are indeed no-ops for systems that have zero-length continuous state and even include a check that the input v/qdot is zero length). Without further evidence, this problem appears to be one at the derived-from-leaf system level.

edrumwri, 2018-07-06:
Per @amcastro-tri, it's not simple/straightforward to reproduce the problem, and I'm not convinced that there is anything happening in the system level that shouldn't be. Will re-open if the issue re-emerges.

closed 2018-07-06 by edrumwri