RobotLocomotion/drake #9107: Bindings shouldn't get added to build_components.bzl

avalenzu, 2018-07-10:
Currently running `build_components_refresh.py` adds new lines for `//bindings/pydrake/*`. It shouldn't (or so I'm told).

EricCousineau-TRI, 2018-07-10:
Shortly after we discussed this, I realized that maybe it should be there, since these headers are available to downstream users in Bazel-land; however, perhaps this shouldn't be merged until https://github.com/RobotLocomotion/drake-shambhala/pull/98 is resolved and merged in.

Either way, I don't think it's harmful either way.

jwnimmer-tri, 2018-11-09:
This is already fixed https://github.com/RobotLocomotion/drake/blob/883d558690c4c1e86a009177ce19dfccd02fac5a/tools/install/libdrake/build_components_refresh.py#L39 (by excluding bindings entirely).  If we want to provide bindings/pydrake utilities helpers to CMake downstreams, that should be a separate issue (and probably not worth worrying about yet.)

closed 2018-11-09 by jwnimmer-tri