RobotLocomotion/drake #9118: MultibodyPlant::CreateDefaultContext()->Clone() throws an exception.

edrumwri, 2018-07-12:
A context created through MultibodyPlant::CreateDefaultContext() throws an exception when its Clone() method is called.

## Reproduction

```
  auto mbp = std::make_unique<MultibodyPlant<double>>();
  const std::string full_name = drake::FindResourceOrThrow(
      "drake/manipulation/models/iiwa_description/sdf/iiwa14_no_collision.sdf");
  multibody::parsing::AddModelFromSdfFile(full_name, mbp.get());
  mbp->Finalize();
  auto context = mbp->CreateDefaultContext();
  auto context_clone = context->Clone();  
```

A stack trace indicates: 

```
Run till exit from #0  drake::systems::Context<double>::Clone (
    this=0x555557a24620)
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/context/drake/systems/framework/context.h:55
abort: Failure at systems/framework/context_base.cc:18 in Clone(): condition 'typeid(source) == typeid(clone)' failed.

Program received signal SIGABRT, Aborted.
0x00007ffff6736428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) back
#0  0x00007ffff6736428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff673802a in __GI_abort () at abort.c:89
#2  0x00005555563f006c in drake::detail::Abort (
    condition=0x55555654ce30 "typeid(source) == typeid(clone)", 
    func=0x55555654ce50 "Clone", 
    file=0x55555654ce56 "systems/framework/context_base.cc", line=18)

```

amcastro-tri, 2018-07-12:
What is the priority of this @edrumwri? do we need to fix this for the ID controller? if so, why?

edrumwri, 2018-07-12:
I could work around it- it's not blocking me- but I would think Clone()'ing
is some critical functionality. Still, I think it's best for you to decide
since I suspect you're buried...

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

amcastro-tri, 2018-07-12:
ok, good. I can always pump its priority up if you need it. I'll always be buried, I just need to prioritize the pile of dirt over my head :-)

PR #14433 (2020-12-09): plant: Add unit test for context cloning
Closes #9118.

Prior to #10581, MbP used a custom context subclass.  I suspect that was a necessary condition for the segfault, but in any case I've now added a regression test that MbP context cloning works as expected.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14433)
closed 2020-12-09 by sherm1