RobotLocomotion/drake #9122: Support fine-grain (geometry level) programmatic collision filtering

SeanCurtis-TRI, 2018-07-12:
## Description

(Described w.r.t. MBP and SG, but true for *any* geometry source).

We have an MBP with multiple bodies. One body (B1) has *multiple* collision geometries (denoted as c₁₁, c₁₂, ..., c₁ₙ). Another body (B2) has collision geometries c₂₁, c₂₂, ..., c₂ₘ. Currently the APIs are well-suited to creating collision filtering between bodies B1 and B2 (i.e., so pairs (c₁₁, c₂₁), (c₁₁, c₂₂), ...,  (c₁ₙ, c₂ₘ) will *not* be considered for collision evaluation. This is done simply by creating two `GeometrySet` instances and adding the `FrameId` for B1 to one set and the `FrameId` for B2 to the other set. (The `FrameId` values imply *all* of the geometries affixed to the frame.) This works well at the *frame*/*body* level. However, if we want to apply collision filtering at a finer level (geometry) there are some obstacles.

`GeometrySet` accepts specific `GeometryId` values as well as `FrameIds`, but the problem is *identifying* the `GeometryId`s. MBP gets populated via parsing an sdf file. So, the geometries are defined and affixed to frames outside of user code. The user code needs some mechanism to recognize which `GeometryId` corresponds to the collision geometry of interest.

## Proposed API

This reflects using the MBP and SG, but the principle should extend to arbitrary geometry sources. It assumes the geometry source can provide a `FrameId` for the frame to which the geometry is affixed.

In this case, we have two bodies: "some_body" and "other_body". "some_body" contains multiple collision geometries, but there are two *specific* geometries we don't want to collide with *any* of the geometries on "some_body" (we'll assume those geometries are named "geometry1" and "geometry2".) So, the goal is to create a collision filter between those two geometries on "some_body" and *all* the geometries on "other_body".

```c++
SceneGraph<double> scene_graph;
MultibodyPlant<double> plant;
// populate plant and scene graph.

// Step 1: Acquire the frame id from the geometry source.
const Body<double>& body = plant.GetBodyByName("some_body");
FrameId frame_id = plant.GetBodyFrameIdOrThrow(body.index());
const Body<double>& other_body =  plant.GetBodyByName("other_body");
FrameId other_frame_id = plant.GetBodyFrameIdOrThrow(other_body.index());

// Step 2: Acquire a QueryObject from the scene graph to introspect the model being built.
const QueryObject<double> query_object = scene_graph.get_model_query_object();

// Step 3: Use the QueryObject to introspect the data we need.
GeometryId geometry_id1 = query_object.GetGeometryIdByNameOrThrow(frame_id, "geometry1");
GeometryId geometry_id2 = query_object.GetGeometryIdByNameOrThrow(frame_id, "geometry2");
GeometrySet set1{{id1, id2}};
GeometrySet set2{other_frame_id};
scene_graph.ExcludeCollisionsBetween(set1, set2);
``` 

## Work to be done

There are several missing pieces of this proposal:

1. __Geometry Names__: There are currently no names associated with geometries in `SceneGraph`. The sdf file *requires* names for `<collision>` and `<visual>` elements and those names must be unique within the scope of the link. Those names should be associated with the geometry in `SceneGraph` to support the function `GetGeometryIdByNameOrThrow()`.
2. __SceneGraph Introspection__: The primary mechanism intended for introspecting `SceneGraph` is the `QueryObject`. However, it's primary use case is to be evaluated from an input port. In this case, we haven't necessarily wired up the diagram and there are no meaningful ports to query. However, geometry/frame registration, collision filter registration, and similar operations invoked on the `SceneGraph` modify the underlying `SceneGraph` *model* (which is eventually copied into the `SceneGraph` context's abstract state). We can provide a query object that can operate on the SceneGraph's model abstract state rather than the state embedded in a context.

@cc @rcory

rcory, 2018-07-12:
This is great! Minor typo, I think, second paragraph under **Proposed API**, "but there are two specific geometries we don't want to collide with any of the geometries on "some_body" ". I believe "some body" should be "other body"?

sherm1, 2018-07-12:
Nice! Should this be a more specific kind of query object, like `ModelQueryObject` or `IntrospectionQueryObject` to distinguish it from the Context-dependent runtime queries?

SeanCurtis-TRI, 2018-07-12:
Maybe, maybe not. I want the `IntrospectionQueryObject` to be fully functionally equivalent to the `QueryObject`. Superficially, they shouldn't be any different. A `QueryObject` gives you the ability to query geometry state. The fact that *which* state you're querying is different is immaterial -- after all, I could have two different contexts and derive a different QueryObject from each one. The existence of one doesn't invalidate the other -- but the *meaning* of the query results is predicated upon my understanding of where the QO came from.

PR #9147 (2018-08-17): Add names to geometries and provide facility to access geometries by name
Adds names to geometries and the ability to get GeometryIds from those names.

Anything *outside* of the `geometry` folder reflects a change to the definition of a `GeometryInstance` -- names are required.  Everything *inside* the `geometry` folder is in support of the geometry names.

Sorry for the size. I can break the inspector out of this PR if you'd prefer a smaller PR.

```
Category            added  modified  removed  
----------------------------------------------
code                372    122       29       
co [... cut]
closed 2018-08-17 by SeanCurtis-TRI