RobotLocomotion/drake #9128: More SceneGraph introspection

SeanCurtis-TRI, 2018-07-12:
## Description

Given a frame id, it would be good to get the geometries  (shape and pose) of the geometries affixed to that pose.

@cc @calderpg-tri

sherm1, 2019-01-16:
@SeanCurtis-TRI @calderpg-tri what is an appropriate priority for this issue?

sherm1, 2019-01-18:
Per f2f discussion this morning with @calderpg-tri I think I understood that this just requires the ability to obtain the generic Shape object, and that @calderpg-tri can provided a reifier to perform specific-shape operations. Is that right? If so can we mark this "ready for work" or "on deck" and assign to @SeanCurtis-TRI or @DamrongGuoy ? Bumping to hi-pri since it sounds both urgent and easy!

SeanCurtis-TRI, 2019-01-18:
Sorry; I have a PR in work. I'm just finalizing the unit tests; it'll go in first thing in the morning.

jwnimmer-tri, 2022-02-01:
There's some name munging still going on, with a TODO citing this issue number, that wasn't removed:

https://github.com/RobotLocomotion/drake/blob/0e2767a06f40e16ceeb670079c6c63a7d2200c45/multibody/plant/multibody_plant.cc#L244-L256

jwnimmer-tri, 2022-02-01:
Maybe \CC @EricCousineau-TRI has input to offer here, as well.

EricCousineau-TRI, 2022-02-01:
At first blush, I'm thinking I wrote the wrong issue number down...

jwnimmer-tri, 2022-02-01:
See also #10406 (which added the TODO).

EricCousineau-TRI, 2022-02-01:
Yup, read through that, but not sure why I mentioned this issue.

The analog issue mention in Python was removed by [#14292; ](https://github.com/RobotLocomotion/drake/pull/14292/files#diff-2797a64a51e7dfb4199df990ec1d6f4191a248a3de6a5cbe188d8c5c548d6e1d)
My guess is, I crossed wires in my head / didn't communicate well about leveraging names in that issue.

Perhaps #14043 is slightly more relevant, but that's only for MultibodyPlant; synchronizing the hierarchy to SceneGraph would be another step.

Options:
- (Simplest) Remove the issue and/or TODO, re-closing this issue.
- (Slightly more work) File new issue about exposing hierarchy in MultibodyPlant to SceneGraph, denoting #14043 as prerequisite. (I have no intent to work on this in the near future.)

jwnimmer-tri, 2022-02-01:
I admit to not fully researching (yet) or understanding the how and why of why the name mangling is happening.

My entry point into this thread was the fact that SDFormat input data with `<collision name="fred">` shows up in Scene Graph with an `Inspector::GetName(GeometryId)` of something other than "fred".  (It currently shows up as "model_instance_name::fred".)

That seems like a bug?  See #16489 at `UndoGetScopedName` where I had to work around it. 

Is the problem here that the `GeometryFrame` name-scoping still needs qualifiers (which is probably fine?), but that the extra `RegisterGeometry` name-scoping is pointless and we should just nuke it right away?

EricCousineau-TRI, 2022-02-01:
Ah, good point.

Then yeah, undoing internal scope mangling and deferring to another mechanism for scoping makes more sense.
How do you envision `GeometryFrame` having qualifiers? (or is it just, "query body from frame, get model instance name, and scope it" ?)

SeanCurtis-TRI, 2022-02-02:
IIRC the only naming requirement SceneGraph has is that sibling geometries (those affixed to the same frame) with the same role must have unique names. MbP performs the prefixing when registering it. I'd been meaning to remove that for ages, but, for the life of me, I can't remember what the limiting factor is. But removing the mangling has my full, enthusiastic suppoprt.

PR #10446 (2019-01-22): Adding Shape introspection
1. SceneGraphInspector and GeometryState provide interface to support querying the specified Shape for a registered geometry.
2. Simple smoke test in SceneGraphInspectorTest
3. Full test of Shape introspection on SceneGraph.

The vast majority of this change is in test.

fixes #9128

```
Category            added  modified  removed  
----------------------------------------------
code                179    0         0        
comments            24     0         0        
blank               20     0         0        
----------------------------------------------
TO [... cut]