RobotLocomotion/drake #9129: Move Check Equilibrium point from `Linearize` to `tilqr`

joycezyu, 2018-07-13:
Per discussion with @hongkai-dai, it probably makes more sense to check if it's an equilibrium point in `tilqr` (`LinearQuadraticRegulator.cc`) instead of in `linear_system.cc`. We've manually disabled locally in order to keep using `Linearize`. Please considering moving equilibrium point check to `tilqr`.

RussTedrake, 2018-07-16:
[FirstOrderTaylorApproximation](http://drake.mit.edu/doxygen_cxx/group__primitive__systems.html#gaa2d7be4940987b73bf532098b2abcb82) is intended for the use case it sounds like you are using?  It is immediately next to `Linearize` in that header file.

I think the check belongs where it is?

cc @hongkai-dai

hongkai-dai, 2018-07-16:
@RussTedrake Yes you are right. I will close this issue.

closed 2018-07-16 by hongkai-dai