RobotLocomotion/drake #9139: Upload checksum and detached signature files for binary releases

jamiesnape, 2018-07-16:


EricCousineau-TRI, 2018-12-06:
Per f2f, this will be worked on shortly.

jamiesnape, 2018-12-13:
Checksum upload was added in https://github.com/RobotLocomotion/drake-ci/commit/66139dd14212318753cba3eb4bd0f1f95d6fcfb5. Need to update the drake documentation, which I will do when the detached signature files are added.

jwnimmer-tri, 2019-09-19:
FYI Whatever ends up here, I would end up reusing for the monthly tagged releases.  Might be worth backlog => low?

jamiesnape, 2019-09-19:
Sure, maybe medium even (low pretty much equals backlog for me at the moment).

EricCousineau-TRI, 2019-09-19:
Setting to "low", with a target date of mid-November. That work?

jwnimmer-tri, 2019-09-19:
Sure, I am in no hurry to sign the releases.  Checksums are OK for now.

jamiesnape, 2021-01-26:
Note, the monthly S3 packages do get signed since I manually upload those (add `.asc` to a given filename and verify with the key at https://drake-packages.csail.mit.edu/drake/release/drake.asc, which is the same key as the apt packages). Basically, to me, this ticket is an appendix to an automated release process. Once that happens, it is additional infrastructure and security.

jwnimmer-tri, 2023-01-18:
All of the binaries on https://github.com/RobotLocomotion/drake/releases/ pages now have sha256 and sha512 files (checksums).

We don't have actualy signatures, but for now I'm ok with the chain of trust going through `https://github`.

closed 2023-01-18 by jwnimmer-tri