RobotLocomotion/drake #9144: Support sentinel identifiers

SeanCurtis-TRI, 2018-07-17:
## Description

[`geometry::Identifier`](https://github.com/RobotLocomotion/drake/blob/master/geometry/identifier.h) provide a type-safe int-like type for using as universal identifiers. As an int-like type, they have a *very* small set of operations possible. To guarantee global uniqueness, they can only be generated via a factory method that guarantees a unique id with every invocation.

However, it would be good to support the concept of sentinel ids. These would be some subset of the ids that have pre-determined semantics. Furthermore, we would like these ids (which can be printed as integers) be defined at compile time so that their values will be fixed regardless of the runtime execution.

For example, `SceneGraph` uses `FrameId` to identify frames registered with the `SceneGraph`. It would be nice if we could define a `FrameId` which represents the world and is fixed across all `SceneGraph` instances.

Creating a static value doesn't do it:

```c++
struct Foo {
  const static FooId magic_id;
};

const FooId Foo::magic_id = FooId::get_new_id();
```
If there were other static duration storage instances of FooId the ordering of these initializations are not guaranteed (so they could end up swapping values).

## Proposal

The `Identifier` class should be receive another template parameter -- an int N reporting the number of reserved sentinel values. It should also include a function that allows retrieval of one of the sentinel values. This would make a sentinel value with a fixed value (determined at compile time) and guaranteed to be constant independent of other compilation/run-time vagaries.

jwnimmer-tri, 2018-07-17:
Except for the requirement of "fixed for all executions no matter what", function-local statics would be great here:
```c++
class Stuff {
 public:
  static FooId get_world_id() {
    static FooId result;
    return result;
  }
};
```
For my money, this is simple and effective, and doesn't require baking some set of tiny integers into the typeid or as a hard-coded constant somewhere.

jwnimmer-tri, 2018-07-17:
@SeanCurtis-TRI yes, exactly.  My point is that I don't think having fixed-for-all-executions is a worthwhile goal.

jwnimmer-tri, 2018-07-17:
The IDs are always a nonce (nonsense).  Serializing the non-world IDs without a translation layer is already doomed.  Labels are not IDs.  Most things like Ground Truth rendering will always need a translation to the consumer's label-int space anyway.

jwnimmer-tri, 2018-07-17:
The debugging use case is valid, but for any single program being debugged, the world nonce will be deterministically chosen.

SeanCurtis-TRI, 2019-07-31:
Closing as I'm persuaded that there's insufficient value in the idea. If anything ever disproves that it can be revisited.

closed 2019-07-31 by SeanCurtis-TRI