RobotLocomotion/drake #9158: Forward kinematics are not thread safe

calderpg-tri, 2018-07-19:
## Description

Expected: Forward kinematics using a thread-local `Context` can be safely parallelized.

```
void CalcAllBodyPosesInWorld(
     const systems::Context<T>& context,
      std::vector<Isometry3<T>>* X_WB) const;
```

What happens: Forward kinematics are not thread-safe.

## Reproduction

If an operation like this is performed

```
#pragma omp parallel for
for (int i = 0; i < 100; i++) {
  const int32_t thread_num = omp_get_thread_num();
  per_thread_plant_context_[thread_num]->get_mutable_state()
      .get_mutable_continuous_state().get_mutable_generalized_position()
          .SetFromVector(q); // Pick any q
  std::vector<Eigen::Isometry3d> X_WB_set;
  plant_->model().CalcAllBodyPosesInWorld(
      *per_thread_plant_context_[thread_num], &X_WB_set);
}
```

the forward kinematics will be incorrect.

## Root cause

Stand-in caching used to implement forward kinematics is not thread safe.

sherm1, 2018-07-19:
A good solution would be to use real caching (inside the thread-local Context) as soon as #9113 lands.

jwnimmer-tri, 2018-07-19:
Even in the meantime, it should be trivial to move the `pc_` and `vc_` storage into the `MultibodyTreeContext`.

amcastro-tri, 2018-07-20:
I'm buried with a pile of things to do and I really don't know when I'll be able to tackle this. Anybody would like to attempt @jwnimmer-tri's approach?, I believe that's the right temporary solution.

edrumwri, 2018-07-21:
I'll work on this.

sherm1, 2018-07-21:
@edrumwri see PR #9167

calderpg-tri, 2018-07-24:
PR [#9167](https://github.com/RobotLocomotion/drake/pull/9167) has landed, so position and velocity kinematics caching is now per-context, not per-MBT/MBP model.

amcastro-tri, 2018-07-25:
nice! can we close this @calderpg-tri? or is it still an issue?

calderpg-tri, 2018-07-25:
I would consider it solved for now. Documentation on how to properly handle multiple parallel queries in their code would probably be helpful to other people in the future.

amcastro-tri, 2018-07-25:
I agree on the docs. If you write them (you are right now the most experienced) I'd volunteer to review :-) 
Closing this now.

closed 2018-07-25 by amcastro-tri