RobotLocomotion/drake #9164: Bug in RigidTransform::GetAsIsometry3()

amcastro-tri, 2018-07-20:
in [rigid_transform.h](https://github.com/RobotLocomotion/drake/blob/master/math/rigid_transform.h#L168) a call to pose.makeAffine(); is missing. Otherwise the fourth row of the isometry has garbage.

mitiguy, 2018-07-24:
Will be fixed with PR 9160.

SeanCurtis-TRI, 2018-08-08:
FTR In my opinion, this was never a bug and shouldn't be done in the translation from `RigidTransfsorm` to `Isometry3`. Because we've used the `Isometry` encoding on the `Eigen::Transform`, `Eigen` makes that last row garbage (or, technically, leaves it as garbage). So, as long as we're going to our aliased `Isometry3` type, the caller has no guarantees that the last row is meaningful, and if they want it to be meaningful, *they* should be obliged to call `makeAffine()` before grabbing the full matrix.

sherm1, 2018-08-08:
I think you're right _technically_ but it feels ugly to me to have the elegant RigidTransform class deliver a row of garbage. I'm glad to have it return the matrix in affine form and let the user make their own garbage later.