RobotLocomotion/drake #9175: Downstream CMake project linker errors with source installation, binary works

5yler, 2018-07-24:
## Description

I'm trying to build a downstream CMake project that links against drake. I've had success on Mac with both a binary installation of drake and a source installation of drake. On Ubuntu 16.04, linking against a binary installation of drake works, whereas using a source installation fails similarly as described in https://github.com/RobotLocomotion/drake/issues/7467 with a linker error: 
```
undefined reference to `RigidBodyTree<double>::getBodyOrFrameName(int) const'
undefined reference to `RigidBodyTree<double>::get_position_name(int) const'
```

## Reproduction

For the source installation, I did the following:
```
git clone git@github.com:MobileManipulation/drake.git
cd drake
bazel build --compiler=gcc-5 //...
mkdir install
bazel run //:install -- $PWD/install
sudo ln -s $PWD/install /opt/drake
```

## Debug Information

Output of `nm` for binary install:
```
18:47 delta /opt/drake/lib nm -D libdrake.so | grep getBodyOrFrameName
0000000000f023f0 W _ZNK13RigidBodyTreeIdE18getBodyOrFrameNameEi
```
Source install:
```
18:48 delta /opt/drake-source/lib nm -D libdrake.so | grep getBodyOrFrameName
000000000122a000 W _ZNK13RigidBodyTreeIdE18getBodyOrFrameNameB5cxx11Ei
```

* What operating system are you running Drake on?
(e.g. Ubuntu 16.04, macOS 10.13)

macOS 10.13.5 (works fine)
Ubuntu 16.04 (nope)

* What version of Bazel and CMake are you using?

```
+ which bazel
/usr/bin/bazel
+ bazel version
Build label: 0.14.1
Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Fri Jun 8 12:17:35 2018 (1528460255)
Build timestamp: 1528460255
Build timestamp as int: 1528460255
+ which cmake
/usr/local/bin/cmake
+ cmake --version
cmake version 3.12.0
```

* If you are using Bazel, please provide the host settings in your present
    project workspace:

```
19:00 delta ~/drake bazel run @drake//common:print_host_settings
INFO: Analysed target @drake//common:print_host_settings (0 packages loaded).
INFO: Found 1 target...
Target @drake//common:print_host_settings up-to-date:
  bazel-bin/external/drake/common/print_host_settings
INFO: Elapsed time: 0.238s, Critical Path: 0.00s
INFO: 0 processes.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
+ capture_cc_env=external/drake/common/capture_cc.env
+ source external/drake/common/capture_cc.env
++ BAZEL_CC=/usr/bin/clang-4.0
+ /usr/bin/clang-4.0 --version
clang version 4.0.0-1ubuntu1~16.04.2 (tags/RELEASE_400/rc1)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```
^ I'm confused about that part - why is the value of `BAZEL_CC` clang after using `--compiler=gcc-5` for the bazel build?

* If you are using CMake, please provide the compilers being used in your
    project:

```
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++
CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-5
CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-5
CMAKE_C_COMPILER:FILEPATH=/usr/bin/cc
CMAKE_C_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-5
CMAKE_C_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-5
```

```
18:56 delta ~ /usr/bin/c++ --version
c++ (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
18:57 delta ~ /usr/bin/cc --version
cc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
```

jwnimmer-tri, 2018-07-25:
```
...
$ bazel build --compiler=gcc-5 //...
...
$ bazel run //:install -- $PWD/install
...
$ bazel run @drake//common:print_host_settings
...
++ BAZEL_CC=/usr/bin/clang-4.0
^ I'm confused about that part - why is the value of BAZEL_CC clang after using --compiler=gcc-5 for the bazel build?
```

Indeed, you noticed a symptom of the root cause.

The `--compiler=gcc-5` flag is not sticky.  If you want to use that flag, you need to pass it to every invocation of `bazel`, including `bazel run //:install` as well as the `print_host_settings`.

This sequence of commands ...
```console
$ bazel build --compiler=gcc-5 //...
$ bazel run //:install -- $PWD/install
```
... actually (1) compiles Drake using GCC, then (2) _re_-compiles Drake using Clang then installs it.

Try passing the `compiler` flag all the time, and let us know if that helps.

I'll also note that in #8881 we are working to make GCC 5 the default on Ubuntu 16.04, but that PR is still in-progress.

5yler, 2018-07-25:
Cool, doing
```
bazel build --compiler=gcc-5 //...
bazel run --compiler=gcc-5 //:install -- $PWD/install
```
worked fine. Thanks!

closed 2018-07-25 by 5yler

jwnimmer-tri, 2018-07-25:
I'll also note for the record that https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_cmake_external is our supported way to use Drake from source from a CMake project.  (Using that mechanism have already used the system GCC as the default compiler.)  That is, you should use Drake's top-level `CMakeLists.txt` to install Drake from source, not run `bazel` directly.  It is not our intention for users to ever `run //:install` manually.  We might make breaking changes to that signature without warning, but we will keep the `CMakeLists.txt` entry point working.