RobotLocomotion/drake #9193: IDM cars break in multilane

agalbachicar, 2018-07-27:
## Description

In PR #9173, where `multilane` is added to `automotive_demo`, adding IDM cars to `multilane_onramp_merge` scenario makes the simulation end with:

```
abort: Failure at bazel-out/k8-dbg/bin/systems/analysis/_virtual_includes/simulator/drake/systems/analysis/simulator.h:613 in StepTo(): condition 'next_sample_time >= step_start_time' failed.
```

A similar issue was reported in #8045 .  

## Reproduction

To reproduce, IDM cars should be added to `automotive_demo` with `multilane_onramp_merge` scenario. 

To run it:

```
bazel run //automotive:demo -- --visualizer --with_multilane_onramp --multilane_num_lanes=3 --num_idm_controlled_maliput_railcar=1 
```

## Debug Information

* What operating system are you running Drake on?

Ubuntu 16.04 from docker.

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?

(a) C++

* Are you (a) building `drake` from source, or (b) from a binary release?

(a) source

sherm1, 2018-07-28:
Some extra info, not sure if it's helpful:
That error message comes from this sequence in Simulator::StepTo():
```c++
    const T step_start_time = context_->get_time();
    // ...
    // How far can we go before we have to take a sampling break?
    const T next_sample_time =
        system_.CalcNextUpdateTime(*context_, timed_events.get());
    DRAKE_DEMAND(next_sample_time >= step_start_time);
```
This indicates that CalcNextUpdateTime() returned a value _earlier_ than the time that was passed in via the context. Or, possibly, one of the values here is NaN likely due to an earlier problem.

@edrumwri may have some further thoughts on how to debug this.

jwnimmer-tri, 2019-06-19:
Per #11603, the automotive folder is deprecated, so I'll close this issue as "won't fix"; @jadecastro if this issue is still important to you, please re-open it in a different issue tracker.

\CC @liangfok FYI

closed 2019-06-19 by jwnimmer-tri