RobotLocomotion/drake #9201: Run a spell checker on CI?

jamiesnape, 2018-07-31:
e.g., [codespell](https://github.com/codespell-project/codespell). Seems we have around 150 files with easily identifiable spelling mistakes at present, per #9200 and upcoming PRs.

jwnimmer-tri, 2020-04-21:
What would the output and/of workflow look like?

jamiesnape, 2020-04-22:
Same workflow as the other linters. Output is along the lines of
```
multibody_doxygen.h:540: homogenous  ==> homogeneous
```
There is a flag for automatically writing changes, so a command line could be written to invoke that .

jwnimmer-tri, 2021-11-10:
The call to action here would be to add `codespell` integration into `drake/tools/workspace` (or maybe as a maintainer-only setup dep, instead) along with brief instructions how to run it in autofix mode.  Then we could have a recurring task to run it every three months (or whatever) and submit a pull request with the fixes.  There is no need to involve CI.

rpoyner-tri, 2021-11-10:
Corrections definitely have to be manual, since the code contains a variety of proper names and neologisms. Some could maybe be kept in a custom dictionary, others maybe not. References to contributor Belinda Teh are an example where dictionaries may not help.

jwnimmer-tri, 2022-05-06:
Closing as "won't fix" -- not a sufficiently valuable use of our time and attention.

closed 2022-05-06 by jwnimmer-tri