RobotLocomotion/drake #9211: Cmake build errors in ubuntu 16.04, matlab 2017b

Jerryxiaoyu, 2018-08-02:
Hi, 
When building with MATLAB R2017b, I got some errors. Coud any one figure it out ?  Thank you.
Error log as the following:
```
-- Not running examples/Wingeron/findTrim because avl was not installed
-- Not running examples/Wingeron/runPassive because avl was not installed
-- Not running examples/grasping/forceClosure/test/testFixedContactsSearchQ1 because lcm was not installed
-- Not running examples/grasping/forceClosure/test/testForceClosureContactsFixedFaces because lcm was not installed
-- Not running examples/grasping/forceClosure/test/testSynthesizeGrasp because lcm was not installed
-- OXYGEN_DIR (oXygen XML Editor) not set. URDF documentation will be excluded from the documentation build.
-- Configuring incomplete, errors occurred!
```
 
The file _CMakeError.log_ as the following:
```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/drl/matlab_drake/drake-last_sha_with_original_matlab/drake-build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_02422/fast"
/usr/bin/make -f CMakeFiles/cmTC_02422.dir/build.make CMakeFiles/cmTC_02422.dir/build
make[1]: Entering directory '/home/drl/matlab_drake/drake-last_sha_with_original_matlab/drake-build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_02422.dir/CheckSymbolExists.c.o
/usr/bin/cc    -fPIE   -o CMakeFiles/cmTC_02422.dir/CheckSymbolExists.c.o   -c /home/drl/matlab_drake/drake-last_sha_with_original_matlab/drake-build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_02422
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_02422.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_02422.dir/CheckSymbolExists.c.o  -o cmTC_02422 
CMakeFiles/cmTC_02422.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_02422.dir/build.make:97: recipe for target 'cmTC_02422' failed
make[1]: *** [cmTC_02422] Error 1
make[1]: Leaving directory '/home/drl/matlab_drake/drake-last_sha_with_original_matlab/drake-build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_02422/fast' failed
make: *** [cmTC_02422/fast] Error 2
```

sherm1, 2018-08-03:
Not sure where to start with this -- assigned to @jamiesnape for a first look but please reassign if necessary.

Jerryxiaoyu, 2018-08-05:
@jamiesnape @sherm1, thanks for your reply. I'm not sure whether this problem is related with Eigen3. I tried to reinstall eigen3, but it still didn't work...

jamiesnape, 2018-08-06:
This is the tag `last_sha_with_original_matlab`?

Jerryxiaoyu, 2018-08-07:
@jamiesnape Yes, that is the tag `last_sha_with_original_matlab`. Some examples I wanted to run have been removed from the master so that I tied to use the previous tag.