RobotLocomotion/drake #9222: Multibody plant SDF parsing / sdformat "preserve unknown elements" is broken

jwnimmer-tri, 2018-08-03:
When I upgrade Drake's commit of SDFormat to the latest upstream `default` (aka master), the "preserve unknown elements" feature causes things to go haywire.  In particular, the new unit test in #9220 fails.

Reverting #9221 will demonstrate this -- 9221 rolls back the SDFormat commit from one that has "preserve unknown", to a month-prior version that doesn't.  If we push forward to a new commit, the unit test will break.

amcastro-tri, 2018-08-03:
From conversation with @jwnimmer-tri, an example failure is:

```
==================== Test output for //multibody/multibody_tree/parsing:multibody_plant_sdf_parser_test:
Using drake_cc_googletest_main.cc

[==========] Running 8 tests from 3 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from AcrobotModelTests
[ RUN      ] AcrobotModelTests.ModelBasics
[       OK ] AcrobotModelTests.ModelBasics (52 ms)
[ RUN      ] AcrobotModelTests.VerifyMassMatrixAgainstBenchmark
[       OK ] AcrobotModelTests.VerifyMassMatrixAgainstBenchmark (67 ms)
[----------] 2 tests from AcrobotModelTests (119 ms total)

[----------] 5 tests from MultibodyPlantSdfParser
[ RUN      ] MultibodyPlantSdfParser.LinkWithVisuals
No HOME defined in the environment. Will not log.
unknown file: Failure
C++ exception with description "From AddModelFromSdfFile():
Error: Plane geometry is missing a <size> child element. Using a size of 1, 1.
Error: Plane geometry is missing a <size> child element. Using a size of 1, 1.
Error: link with name[link3] already exists.
" thrown in the tesINFO: From Testing //multibody/multibody_tree/parsing:multibody_plant_sdf_parser_test:
t body.
[  FAILED  ] MultibodyPlantSdfParser.LinkWithVisuals (59 ms)
```

jwnimmer-tri, 2018-08-03:
The links posted in #9220 show all of the failures (that the unit tests exposes).

amcastro-tri, 2018-08-03:
@SeanCurtis-TRI was asking me why the "high priority" @jwnimmer-tri. Could you elaborate?

jwnimmer-tri, 2018-08-03:
Yes, good question.  When we pin an external to an older revision because their master is broken with respect to us, it means that we can _never_ upgrade to any newer version, for any bugfix, new feature, security hole, or any other cause where we'd want to be able to pull in new code from upstream.  In short, we are "frozen" at an old revision of sdformat, and can't unfreeze until this issue is resolve.  To me, that seems like a high-priority problem.  Otherwise, all parsing-related development or bugfixes are blocked indefinitely.

sherm1, 2018-08-03:
FYI @nkoenig (we're not sure what's wrong yet)

nkoenig, 2018-08-09:
I'll take a look at this problem.

jwnimmer-tri, 2018-08-31:
\CC @EricCousineau-TRI FYI

scpeters, 2018-09-01:
I've been able to reproduce the problem, and I've been testing a branch from @nkoenig that appears to fix it:

https://github.com/RobotLocomotion/drake/compare/master...nkoenig:sdf_include_tags_update

I'll try rebasing it and submitting a pull request

sherm1, 2018-09-01:
Thanks, @scpeters!

scpeters, 2018-09-05:
I just finished rebasing the branch (it was a bit tricky), and I'll test it tomorrow.

* https://github.com/RobotLocomotion/drake/compare/master...scpeters:sdf_include_tags_update?expand=1

scpeters, 2018-09-05:
upstream fix: https://bitbucket.org/osrf/sdformat/pull-requests/462/parser-dont-duplicate-elements-when/diff

scpeters, 2018-09-05:
and it's merged upstream https://bitbucket.org/osrf/sdformat/commits/2fed80e6bc44afe6efc3ff71a67e14defb96cce5

scpeters, 2018-09-05:
@jwnimmer-tri is that enough or do you want me to update the bazel workspace too?

jwnimmer-tri, 2018-09-05:
Nah, I think @amcastro-tri or @SeanCurtis-TRI can upgrade Drake to use that commit.

amcastro-tri, 2018-09-06:
cc'ing @EricCousineau-TRI, I believe you wanted to use this right away.

EricCousineau-TRI, 2018-12-10:
Sorry for the late follow-up, but it's unclear who is responsible for the next step in closing this issue. I will briefly check to see if our current version is still affected by this, but if not, we will need to (a) lower the priority unless that's inappropriate, and (b) find one owner to be responsible for fixing this.

PR #10181 (2018-12-11): sdformat: Upgrade to latest upstream
Resolves #9222

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10181)
closed 2018-12-11 by jwnimmer-tri