RobotLocomotion/drake #9238: Bazel (and cmake) compilation fails

gstavrinos, 2018-08-07:
Ubuntu 18.04, after installing all the prerequisites using the script provided, I get both with bazel and cmake the following error:

`ERROR: /home/gstavrinos/drake/geometry/BUILD.bazel:55:1: C++ compilation of rule '//geometry:frame_kinematics' failed (Exit 1)
In file included from bazel-out/k8-opt/bin/common/_virtual_includes/autodiff/drake/common/autodiff.h:31:0,
                 from geometry/frame_kinematics_vector.cc:8:
bazel-out/k8-opt/bin/common/_virtual_includes/autodiff/drake/common/autodiffxd.h: In constructor 'drake::geometry::FrameKinematicsVector<KinematicsValue>::FrameKinematicsVector(drake::geometry::SourceId, const std::vector<drake::geometry::Identifier<drake::geometry::FrameTag> >&) [with KinematicsValue = Eigen::Transform<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1> >, 3, 1, 0>]':
bazel-out/k8-opt/bin/common/_virtual_includes/autodiff/drake/common/autodiffxd.h:103:26: error: '*((void*)&<anonymous> +16)' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     m_value = other.value();
`

soonho-tri, 2018-08-07:
FYI, this problem was reported in https://github.com/RobotLocomotion/drake/pull/9216. AFAIK, @SeanCurtis-TRI is working on a fix.

gstavrinos, 2018-08-07:
Thanks for the prompt reply. I search for something similar, but github search was not helpful. I will keep an eye on that issue!

jwnimmer-tri, 2018-08-07:
Duplicate of #9233.

closed 2018-08-07 by jwnimmer-tri