RobotLocomotion/drake #9241: Implicit Stribeck Solver convergence problem

sherm1, 2018-08-07:
@rcory has a model "simple gripper with 'thin box'". He applied a 15N force to the gripper and observed non-physical oscillations in the contact-normal direction. Here are several experiments he did, any of which makes the oscillations go away:
- apply 20N instead
- cut the step size from 1e-3 to 1e-4
- doubled the penetration allowance from 1e-6 to 2e-6

(That's why we're confident these are non-physical oscillations.) I surmised that the problem might be premature convergence in the implicit stage of the solver. That solver uses `tol=relative_tolerance*stiction_tolerance`, then calculates a 2-norm of the tangential velocity error (`vt_error`) and terminates if `vt_error < tol`. The relative_tolerance defaults to 1e-2 and Rick had set the stiction_tolerance to 1e-3, so tol=1e-5. And that only depends on tangential velocities, but Rick's oscillation problem is in the normal direction, and since there is no gravity in this problem it doesn't appear that there is any tangential velocity at all.

I suggested reducing relative_tolerance to force the solver to converge tighter. That fixed the problem, but not until we got relative_tolerance down to 1e-8, meaning tol=1e-11!  That seems much too tight, but it is understandable since we actually need to converge the velocities in the normal direction, but there is no control on that. 

This makes me think that the convergence criteria in the implicit Stribeck solver may need to be reconsidered. 

/cc @amcastro-tri @edrumwri

amcastro-tri, 2018-08-08:
confirmed on this end. If I force the minimum number of iterations to 10 (way too high for most cases) the problem goes away. Clearly the NR iteration is terminating earlier than it should. I'll explore using an additional convergence criteria.

amcastro-tri, 2018-08-08:
FYI. Monitoring convergence of the normal separation velocities (in addition to the monitoring of tangential velocities) does fix the problem nicely. 
I'd just need to come up with a nice self contained unit test for a PR, ideas?

sherm1, 2018-08-08:
I wonder if this could be reproduced with just a stiff frictionless ball with gravity, resting on the ground or bouncing a little? Should converge prematurely if the normal velocities aren't included in convergence.

amcastro-tri, 2018-08-08:
I'll reopen until we push a more stringent unit test (one that fails before #9244 and does run after the fix).

sherm1, 2020-05-11:
@amcastro-tri can we close this?

amcastro-tri, 2020-05-13:
The fix was pushed long ago. Closing.

PR #9244 (2018-08-08): Monitor normal separation velocities
This PR fixes #9241. It introduces monitoring of normal velocities convergence which will take priority for cases with low or zero tangential velocities.

Unit tests in master should be enough to validate this. However, we will push a very specific unit test for this functionality as soon as possible in a separate PR.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9244)
closed 2020-05-13 by amcastro-tri