RobotLocomotion/drake #9252: Add `get_num_total_states()` to system.h

hongkai-dai, 2018-08-09:
In `system.h`, we have function `get_num_total_inputs()` and `get_num_total_outputs()`, one might ask why we cannot have a function `get_num_total_states()` (or `get_num_total_continuous_states`, `get_num_total_discrete_states()`). Currently I have to call `AllocateContext()`, and then get the number of states through the context. 

@sherm1

hongkai-dai, 2018-09-05:
One classical example on a diagram with multiple leaf system states, is a system with a state observer. The system is in this form

- `ẋ = A * x + B * u` is the system to be controlled.
- `y = C * x` is the output of the system.
- An observer system to estimate the state of the system, with the estimated state x_bar
  `ẋ_bar  = A * x_bar + B * u + L*(y - C * x_bar)`

The diagram contains both the original system `ẋ = A * x + B * u`, together with another leaf system as the observer. The total continuous state of the system should be `x` and `x_bar`.

You might want to consider this example in the test.

DamrongGuoy, 2018-09-05:
As expected, adding this pure virtual function (`virtual ...=0`) into `System`:
```
class System {
    virtual int get_num_continuous_states()  /**/const =0;/**/
```
caused compile errors in unit tests (`TestSystem, ValueIOTestSystem, ComputationTestSystem`) because they became abstract classes.

I'll fix them by adding a fictitious `final` function into those unit tests like this:
```
class TestSystem /* and ValueIOTestSystem and ComputationTestSystem */ {
  int get_num_continuous_states() /**/final/**/ const /**/{ return 0; }/**/
```

DamrongGuoy, 2018-09-06:
Actually, after reading the code a bit more, I think these are the number of continuous state variables for these unit tests:
```
TestSystem 0
ValueIOTestSystem 0
ComputationTestSystem 3 q(1) v(1) z(1)
```
Certainly the one suggested by @hongkai-dai  (system with a state observer) would be more relevant.

sherm1, 2018-09-06:
For stubs that aren't intended to be called you can use the `DRAKE_ABORT()` macro to make that clear, e.g.
```c++
int get_num_continuous_states() const  final { DRAKE_ABORT(); }
```