RobotLocomotion/drake #9263: Error in building drake from source via Bazel in Ubuntu 18.04 Bionic Beaver

pvonmoradi, 2018-08-14:
## Description
I have installed all packages by running the `.sh` file for my distro. I `cd`ed into root of drake and executed `bazel build //...`. It took a very very long time (~5h) for my machine (AMD E1 CPU) to do the build and compile operations. The procedure was interrupted by an error. Here is the dump:
```
pooya@pooya-Lenovo-ideapad-110-15ACL:~/drake$ bazel build //...
DEBUG: /home/pooya/.cache/bazel/_bazel_pooya/5e3236f9a8255a1afbae8329dcfb9f4b/external/bazel_tools/tools/build_defs/repo/http.bzl:48:5: ctx.attr.build_file @drake//tools/workspace/eigen:package.BUILD.bazel, path /home/pooya/.cache/bazel/_bazel_pooya/5e3236f9a8255a1afbae8329dcfb9f4b/external/drake/tools/workspace/eigen/package.BUILD.bazel
DEBUG: /home/pooya/.cache/bazel/_bazel_pooya/5e3236f9a8255a1afbae8329dcfb9f4b/external/bazel_tools/tools/build_defs/repo/http.bzl:48:5: ctx.attr.build_file @drake//tools/workspace/ignition_math:package.BUILD.bazel, path /home/pooya/.cache/bazel/_bazel_pooya/5e3236f9a8255a1afbae8329dcfb9f4b/external/drake/tools/workspace/ignition_math/package.BUILD.bazel
DEBUG: /home/pooya/.cache/bazel/_bazel_pooya/5e3236f9a8255a1afbae8329dcfb9f4b/external/bazel_tools/tools/build_defs/repo/http.bzl:48:5: ctx.attr.build_file @drake//tools/workspace/ignition_rndf:package.BUILD.bazel, path /home/pooya/.cache/bazel/_bazel_pooya/5e3236f9a8255a1afbae8329dcfb9f4b/external/drake/tools/workspace/ignition_rndf/package.BUILD.bazel
INFO: Analysed 6482 targets (0 packages loaded).
INFO: Found 6482 targets...
INFO: From Compiling systems/trajectory_optimization/direct_transcription.cc:
In file included from bazel-out/k8-opt/bin/common/_virtual_includes/essential/drake/common/drake_throw.h:5:0,
                 from bazel-out/k8-opt/bin/systems/framework/_virtual_includes/context/drake/systems/framework/context.h:7,
                 from bazel-out/k8-opt/bin/systems/trajectory_optimization/_virtual_includes/direct_transcription/drake/systems/trajectory_optimization/direct_transcription.h:6,
                 from systems/trajectory_optimization/direct_transcription.cc:1:
bazel-out/k8-opt/bin/systems/trajectory_optimization/_virtual_includes/multiple_shooting/drake/systems/trajectory_optimization/multiple_shooting.h: In member function 'void drake::systems::trajectory_optimization::DirectTranscription::ConstrainEqualInputAtFinalTwoTimesteps()':
bazel-out/k8-opt/bin/systems/trajectory_optimization/_virtual_includes/multiple_shooting/drake/systems/trajectory_optimization/multiple_shooting.h:109:29: warning: assuming signed overflow does not occur when assuming that (X - c) <= X is always true [-Wstrict-overflow]
     DRAKE_DEMAND(index >= 0 && index < N_);
                  ~~~~~~~~~~~^~~~~~~~~~~~
bazel-out/k8-opt/bin/common/_virtual_includes/essential/drake/common/drake_assert.h:107:26: note: in definition of macro 'DRAKE_DEMAND'
     if (!Trait::Evaluate(condition)) {                                       \
                          ^~~~~~~~~
INFO: From Compiling external/lcm/lcm-logger/lcm_logger.c:
external/lcm/lcm-logger/lcm_logger.c: In function 'open_logfile':
external/lcm/lcm-logger/lcm_logger.c:150:64: warning: '%02d' directive output may be truncated writing between 2 and 11 bytes into a region of size between 0 and 4095 [-Wformat-truncation=]
             snprintf(logger->fname, sizeof(logger->fname), "%s.%02d", logger->fname_prefix,
                                                                ^~~~
In file included from /usr/include/stdio.h:862:0,
                 from external/lcm/lcm-logger/lcm_logger.c:4:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: '__builtin_snprintf' output between 4 and 4108 bytes into a destination of size 4096
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
external/lcm/lcm-logger/lcm_logger.c:155:59: warning: '.0' directive output may be truncated writing 2 bytes into a region of size between 1 and 4096 [-Wformat-truncation=]
         snprintf(logger->fname, sizeof(logger->fname), "%s.0", logger->fname_prefix);
                                                           ^~
In file included from /usr/include/stdio.h:862:0,
                 from external/lcm/lcm-logger/lcm_logger.c:4:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: '__builtin_snprintf' output between 3 and 4098 bytes into a destination of size 4096
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2529s linux-sand[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2530s linux-sand[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2531s linux-sand[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2532s linux-sand[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2533s linux-sand[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2533s linux-sand[1,898 / 1,962] 2 actions running
    Compiling bindings/pydrake/systems/framework_py_systems.cc; 2534s linux-sand[1,898 / 1,962] 2 actions running
ERROR: /home/pooya/drake/geometry/BUILD.bazel:55:1: C++ compilation of rule '//geometry:frame_kinematics' failed (Exit 1)
In file included from bazel-out/k8-opt/bin/common/_virtual_includes/autodiff/drake/common/autodiff.h:31:0,
                 from geometry/frame_kinematics_vector.cc:8:
bazel-out/k8-opt/bin/common/_virtual_includes/autodiff/drake/common/autodiffxd.h: In constructor 'drake::geometry::FrameKinematicsVector<KinematicsValue>::FrameKinematicsVector(drake::geometry::SourceId, const std::vector<drake::geometry::Identifier<drake::geometry::FrameTag> >&) [with KinematicsValue = Eigen::Transform<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1> >, 3, 1, 0>]':
bazel-out/k8-opt/bin/common/_virtual_includes/autodiff/drake/common/autodiffxd.h:103:26: error: '*((void*)&<anonymous> +16)' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     m_value = other.value();
               ~~~~~~~~~~~^~
cc1plus: some warnings being treated as errors
INFO: Elapsed time: 12784.403s, Critical Path: 3661.92s
INFO: 687 processes: 653 linux-sandbox, 29 local, 5 worker.
FAILED: Build did NOT complete successfully
```
# Reproduction
`~/drake$ bazel build //...`
* What operating system are you running Drake on?
Ubuntu 18.04: Bionic-Beaver

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?
I guess C++

* Are you (a) building `drake` from source, or (b) from a binary release?
Source
* What version of `python` are you using?
```
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
```
* If you are building from source or using the `drake` C++ library as an
external:

    * What version of Bazel and CMake are you using?
```
/usr/bin/bazel
Build label: 0.14.1
Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Fri Jun 8 12:17:35 2018 (1528460255)
Build timestamp: 1528460255
Build timestamp as int: 1528460255
/usr/bin/cmake
cmake version 3.10.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).

```
    * If you are using Bazel, please provide the host settings in your present
    project workspace:

```
  bazel-bin/external/drake/tools/cc_toolchain/print_host_settings
INFO: Elapsed time: 78.438s, Critical Path: 0.58s
INFO: 2 processes: 1 linux-sandbox, 1 local.
INFO: Build completed successfully, 5 total actions
INFO: Build completed successfully, 5 total actions
+ capture_cc_env=external/drake/tools/cc_toolchain/capture_cc.env
+ source external/drake/tools/cc_toolchain/capture_cc.env
++ BAZEL_CC=/usr/bin/gcc
+ /usr/bin/gcc --version
gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

* If you are building `drake`
[from source](https://drake.mit.edu/from_source.html):

    * What git commit of `drake` are you using?

        > bdc0e0713

    * Are you building `drake` with Bazel or CMake?

        > Bazel

    * Are you using `drake` directly, or consuming it as an external?

        > directory

pvonmoradi, 2018-08-14:
**1-** I guess my issue is realated to this https://github.com/RobotLocomotion/drake/issues/9238
**2-** What are those DEBUG and INFO tags? Are there any other errors in my build process?
**3-** I couldn't find somewhere else to ask this, but I've also installed a docker based drake on my other machine (Win7, Core i7-720 a rather old cpu). Now when I ran `double_pendulum` simulation, it rendered very slow throw x-window. Does building from source cause a faster runtime of simulations w.r.t docker-based solution?

sherm1, 2018-08-17:
I believe this is a duplicate of #9233 and was fixed by #9259 which has merged. Please update to the latest Drake master and re-file if you are still having a problem.

closed 2018-08-17 by sherm1