RobotLocomotion/drake #9264: `//:py/install` has either become slow or IO on CI is flaky with `-c dbg`

EricCousineau-TRI, 2018-08-14:
Several PRs and Continuous build have been failing on this test:
- https://drake-cdash.csail.mit.edu/testSummary.php?project=1&name=//:py/install_test&date=2018-08-14
- https://github.com/RobotLocomotion/drake/pull/9254
     - https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-experimental-debug/10332/
- https://github.com/RobotLocomotion/drake/pull/9260
     - https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-bazel-experimental-debug/10333/

All of the above tests appear to be failing during the installation step, which tends to point to either an IO issue on CI, or possibly because our debug binary sizes have increased to such a point to cause further issues.

I'd suggest we wait for a bit longer to see if the flakiness subsides; if it does not, we should consider increasing the timeout for this test.

\cc @sherm1 @avalenzu @fbudin69500 @jwnimmer-tri

sherm1, 2018-08-14:
I restarted the Debug build in #9260 and py_install did not time out that time.

EricCousineau-TRI, 2018-08-14:
Yeah, I had restarted one of the Continuous builds and it worked; trying that out for Andres's PR.
An alternative is to mark this test as `flaky` if it continues, but that has a slightly funny smell

jamiesnape, 2018-08-15:
No reason CI should be particularly flaky. The test will probably run quicker when the EBS volume is warm, but otherwise it should be reasonably consistent. You can check if the EBS volume is warm or cold by looking for a console message near the start of the output. Flaky is most likely just going to cause the same result to occur three times since each run would write to a new part of the disk.

PR #9265 (2018-08-15): install: Increase test timeout
Resolves #9264.

Per the issue, perhaps we wait to see if flakiness subsides; otherwise, we increase just the timeout (not the resources, unless that would be useful).

\cc @fbudin69500 @jamiesnape

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9265)
closed 2018-08-15 by sammy-tri