RobotLocomotion/drake #9267: Protobuf version issue

ash22194, 2018-08-15:
> Please fill out all quoted sections (`> `) that need a response; feel free
to remove the quoted sections that do not need a response.

> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
Yes
> * Have you reviewed existing GitHub issues and StackOverflow posts?
Yes
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?
Yes

## Description

> Please provide a short description: expected behavior, and what actually
happens. If this is an enhancement request, please provide a concrete use case.

Library versions should be specified in binary_distribution/Brewfile. Drake compiles well with protobuf version 3.1 but fails to compile with version 3.6. Such issues may arise with other libraries as well. A simple fix would be enforcing version check while installing dependencies.

## Reproduction

> Please provide a minimal snippet of code or shell commands (<10 lines)
that reproduces the issue. If it is self-contained, but 10-50 lines,
please link to a Gist.

> If your changes to `drake` are more comprehensive, please try to isolate your
problem to the above amount, ensuring that you are using a branch as close to
`master` as possible.

## Debug Information

> If this is an enhancement / feature request that is not platform-specific,
you do not need to include this section. However, please ensure that the
current version of `master` does not already provide what you want.

* What operating system are you running Drake on?
(e.g. Ubuntu 16.04, macOS 10.13)

    > macOS 10.13

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?

    > (c)

* Are you (a) building `drake` from source, or (b) from a binary release?

    > (a) from source

* What version of `python` are you using?

    > Python 2.7.15

* If you are building from source or using the `drake` C++ library as an
external:

    * What version of Bazel and CMake are you using?

        > Build label: 0.15.2-homebrew
           cmake version 3.12.1

    * If you are using Bazel, please provide the host settings in your present
    project workspace:

        > Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.1.0 (clang-902.0.39.2)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

    * If you are using CMake, please provide the compilers being used in your
    project:

        > Apple LLVM version 9.1.0 (clang-902.0.39.2)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

* If you are building `drake`
[from source](https://drake.mit.edu/from_source.html):

    * What git commit of `drake` are you using?

        > 0db9fbe

    * Are you building `drake` with Bazel or CMake?

        > Bazel

    * Are you using `drake` directly, or consuming it as an external?

        > directly

    * If an external, is it most like
    [`drake_bazel_external`](https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_bazel_external) or
    [`drake_cmake_external`](https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_cmake_external)?

        > Replace this with your answer.

* If you are using `drake` [from a binary release](https://drake.mit.edu/from_binary.html):

    * What are the contents of `.../drake/share/doc/drake/VERSION.txt`?

        > 20180815113952 0db9fbe969ee462fd7fdb51dc414e838241b37ff

    * What URL did you download `drake` from?

        > https://github.com/RobotLocomotion/drake

    * Are you writing a CMake project which is using `drake`, like
    [`drake_cmake_installed`](https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_cmake_installed)?

        > No

sherm1, 2018-08-17:
@jamiesnape please reassign if you are the wrong person for this.

jamiesnape, 2018-08-20:
> Library versions should be specified in binary_distribution/Brewfile

No, from the documentation of `homebrew-bundle`:

> Homebrew does not support installing specific versions of a library, only the most recent one, so there is no good mechanism for storing installed versions in a .lock file.

> Drake compiles well with protobuf version 3.1 but fails to compile with version 3.6.

No, we test on CI with Protobuf 3.6.0 multiple times per day on multiple Mac platforms in multiple configurations of Drake, so I am confident there is no inherent issue with Drake and Protobuf 3.6. If you have specific error, please detail and feel free to reopen.

closed 2018-08-20 by jamiesnape