RobotLocomotion/drake #9269: Error in RgbdCamera constructor -- renderer viewpoint not getting set

SeanCurtis-TRI, 2018-08-15:
When invoking [the constructor](https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/rgbd_camera.cc#L119) which takes a renderer for a *fixed* camera, the renderer's viewpoint does not get set. 

There should be an invocation on the renderer's `UpdateVIewpoint` method. There is only one, in [`ResetRenderer()`](https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/rgbd_camera.h#L268).

I propose that it should simply be added to `InitRenderer()` which gets invoked on the renderer whether it is instantiated internally or externally. 

Testing the correctness of this is challenging because there is no public API to examine the renderer's viewpoint matrix.

kunimatsu-tri, 2018-08-15:
I reassigned this to @thduynguyen since this constructor is introduced by him at this PR: https://github.com/RobotLocomotion/drake/pull/9102

SeanCurtis-TRI, 2018-08-16:
Sounds good.

jwnimmer-tri, 2019-07-02:
@SeanCurtis-TRI I think this issue refers to the attic camera; as such, can we WONTFIX it?

SeanCurtis-TRI, 2019-07-02:
Suits me.

closed 2019-07-02 by jwnimmer-tri