RobotLocomotion/drake #9274: RgbdRendererOspray::ImplRgisterVisual() -- only allows for a single geometry

SeanCurtis-TRI, 2018-08-16:
## Description

If multiple geometries are assigned to an `RgbdRendererOspray` only the *last* geometry will appear.

## Solution

`RgbdRendererOspray::Impl` has [two members](https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/rgbd_renderer_ospray.cc#L154):
  - `std::array<vtkNew<vtkActor>, kNumOutputImage> actors_;`
  - `std::array<vtkNew<vtkPolyDataMapper>, kNumOutputImage> mappers_;`
For each registered visual, it assigns geometry to the common actors array and connects them to the common mappers.

If you contrast that with the implementation in `RgbdRendererVTK::Impl` you'll note that each invocation of `ImplRegisterVisual()` [instantiates a unique set of actors and mappers](https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/rgbd_renderer_vtk.cc#L421).

`RgbdRendererOspray` should mirror this behavior. When it is modified to match, then all registered geometries appear.

jamiesnape, 2018-08-20:
If I recall correctly, there are non-deterministic segfaults if the members in `RgbdRendererOspray::Impl` are changed to locals.

EricCousineau-TRI, 2018-08-20:
Yes, that is correct.

If you look at the diff between `r11` and `r12` on this file:
https://reviewable.io/reviews/robotlocomotion/drake/8939/systems/sensors/rgbd_renderer_ospray.cc
you'll note the shift from using locals to using member values.

SeanCurtis-TRI, 2018-08-20:
I'm running it with locals and haven't had a single geometry-based segfault. In talking to @kunimatsu-tri the current working hypothesis is that it may be an artifact of its interaction with the test harness.

kunimatsu-tri, 2018-08-20:
When I run VTK-OSPRay program, regardless it's one in drake or stand alone cmake program, I get tons of `valgrind` errors. 

Most of them are:
- `Conditional jump or move depends on uninitialised value(s)`
- `Use of uninitialised value of size 8`
 from libospray_module_ispc.so.1.5.0.

And some of them are:
-  Invalid read of size 4
 at destruction time and related to `embree.so.2.17.4` and `ospray::api::ISPCDevice`.

I'm digging more and will report if I find something related.

kunimatsu-tri, 2018-08-20:
I reverted what @EricCousineau-TRI pointed out and try running gtest.

One observation I think related so far is that if I leave only `MeshTest` in `rgbd_renderer_ospray_test.cc` and remove all the other tests, it runs no problem. I repeatedly ran 100 times for that test, and get zero unexpected termination which I had before by running all the test.

If I add one more test, say `CylinderTest`, back to `rgbd_renderer_sopray_test.cc`, and repeatedly run 100 times for that test, I get unexpected termination. 

I still don't know why this happens.

EricCousineau-TRI, 2018-12-06:
Placed the priority at medium given the activity surrounding this; lemme know if we want to change it.

jamiesnape, 2019-02-26:
This is really coupled with #8978, so maybe this should be assigned to me also?

EricCousineau-TRI, 2019-02-26:
Sounds good to me! (I'm sure @kunimatsu-tri won't complain :P)

SeanCurtis-TRI, 2019-02-26:
I agree. It looks like this reported issue is the product of the proposed solution to the issues encountered in the original PR. Putting these issues into a single basket makes sense to me.

EricCousineau-TRI, 2019-03-07:
(FTR Resolution is coupled to #8978 b/c we want to de-singleton these bits.)

SeanCurtis-TRI, 2020-09-25:
Closing this as an `attic` artifact that no longer matters.

closed 2020-09-25 by SeanCurtis-TRI