RobotLocomotion/drake #9279: Cannot make a system symbolic inspector for a system that is instantiated with symbolic::Expression

hongkai-dai, 2018-08-17:
If we declare a templated system
```C++
template <typename T>
class ToySystem1 : public leaf_system<T> {
};
```
then we can construct its `SystemSymbolicInspector`. But if the system is is instantiated with `symbolic::Expression` as
```C++
class ToySystem2 : public leaf_system<symbolic::Expression> {
}
```
Then when I call `MakeSystemSymbolicInspector()`, I got a nullptr.

The problem is in https://github.com/RobotLocomotion/drake/blob/master/systems/framework/system.h#L1347 to convert this system to scalar type `symbolic::Expression`. `ToSymbolicMaybe` thinks it is not convertible.

But the system `ToySystem2` is already instantiated with `symbolic::Expression`, so `ToSymbolicMaybe()` should be a no-op.

cc @sherm1 @soonho-tri @jwnimmer-tri

jwnimmer-tri, 2018-08-17:
I guess I never considered that a System would support _only_ the symbolic scalar type, and not support `double`.  Is that a thing we want to allow for?

hongkai-dai, 2018-08-17:
When we do system analysis, we may only consider system with symbolic scalar type.

jwnimmer-tri, 2018-08-17:
I can look at this in a few weeks later or so.  In the meantime if you need a fix, you could have someone else take a look.

It is possible that this even would work right now, if you follow the http://drake.mit.edu/doxygen_cxx/group__system__scalar__conversion.html instructions and pass a `SystemTypeTag` to the `LeafSystem` constructor, and specialize the `scalar_conversion::Traits` to only allow for `Expression`-to-`Expression` transmogrification.

jwnimmer-tri, 2018-08-17:
Actually, the easy work-around for now, is to support both `double` and `Expression` and have the `ToySystem` be templated on `T`.  We have plenty of example of that, which work fine.

hongkai-dai, 2018-08-17:
This is not urgent. I do not need this working right now, and as you mentioned, I can have easy work around to templatize the system with type `T`.

jwnimmer-tri, 2018-08-28:
Actually, the work-around doesn't work, either.  If you ask a templated `MySystem<Expression>` for its sparsity, it does _not_ use the symbolic inspector!  I will submit a fix today for both that defect, and the original issue (untemplated).

jwnimmer-tri, 2018-08-28:
... and fixing this breaks VectorSystem, because its Calc method asks about feedthrough during computation.  I'll fix that too.

PR #9321 (2018-08-29): framework: Improve symbolic sparsity handling
Closes #9279.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9321)
closed 2018-08-29 by jwnimmer-tri