RobotLocomotion/drake #9285: drake::multibody::parsing::AddModelFromUrdfFile

RussTedrake, 2018-08-21:
I would like to request that we provide the API AddModelFromUrdfFile for MultiBodyPlant/SceneGraph.  I *believe* that now given #8718, the most natural implementation would be passing a trivial sdf file with a single line to include the urdf filename string.  

This will help me port the existing examples over to MBP/SG

sherm1, 2018-08-21:
@amcastro-tri -- @RussTedrake says this would unblock him if it is possible for a quick fix.

amcastro-tri, 2018-08-21:
@avalenzu, I believe you tried this before? Did you have some level of success with this approach?

avalenzu, 2018-08-21:
I had success in that I ended up with a model, but some of the joint axes were wrong.

RussTedrake, 2018-08-21:
sitting next to @avalenzu now.  i think we both agree that we should add the capability, but then send bug reports to sdformat when we find errors (as opposed to waiting passively for sdformat to get better).  Perhaps we need to put a warning on the method documentation, too?

sammy-tri, 2018-08-21:
Pinging @jwnimmer-tri (who is currently on vacation) and who I've talked to about this a number of times.  I can't remember his position clearly so I won't do him the disservice of stating it incorrectly here.

If we go the sdformat route, we need to prepare for the eventuality that we will need to devote developer resources to improving sdformat's URDF (and `<include>`) handling.  It's tough for me to guess how much effort this would be.

I've taken a brief look into porting drake's existing RBT-based URDF parser over to MBP and I think it might be worth considering.  It's not that much code (a few hundred lines) if you ignore some of the quirks of the external API which we don't immediately need to support and I think the changes would be mostly formulaic.  It also seems more likely to preserve existing behavior for RBT based examples.  I haven't had an opportunity to actually try it and probably wouldn't for a few more weeks unfortunately.

scpeters, 2018-08-22:
I'll share my thoughts until @nkoenig gets back. Currently sdformat converts URDF files to an SDF version 1.4 and then converts forward to 1.6 (see [issue 187](https://bitbucket.org/osrf/sdformat/issues/187) and [issue 189](https://bitbucket.org/osrf/sdformat/issues/189/convert-urdf-files-to-sdf-with)). This is fine for the purposes of loading a model for simulation, but it loses the tree structure inherent to URDF, which is a bummer if you wanted to use it for control.

We do have some forthcoming plans to improve sdformat to allow conversion back to URDF. Part of this is based on the [frame attribute to the pose element](https://bitbucket.org/osrf/sdformat/src/5eac19ab542539b1f0de63905d45a225e54651bc/sdf/1.6/pose.sdf#lines-5:7) in sdf 1.5 (though it's not widely used yet). The idea is that a tree structure could be encoded in an SDF by specifying parent frames by name to override the default SDF frame hierarchy (model -> link -> collision). There is some [design documentation here](https://osrfoundation.atlassian.net/wiki/spaces/GAZ/pages/35740/Gazebo+X#GazeboX-CoordinateFrameRepresentation), but we haven't started implementing it yet.

If you have any more detail about your lack of faith in current sdformat urdf parsing, I can look into those concerns too.

jwnimmer-tri, 2018-08-27:
> ...  sdf file with a single line to include the urdf filename string.

IIRC (per @sammy-tri chat a while ago -- correct me if I'm wrong), you don't even need to do this -- if you ask sdformat to open the URDF, it just does it for you (using the transformations mentioned above) and gives you the parse tree that we already consume.  The method requested here could just delegate to "AddModelFromSdfFile" under the hood for now, and it does something (to the extent that sdformat's behavior is what we want).

---

On the bigger question, I haven't really dug into what we should be doing with parsing, but my gut feeling right now is that @sammy-tri is correct -- that we should port Drake's URDF parser for use with multibody, not use sdformat.  (Honestly, I'm not sure that we should use sdformat for SDF parsing, either.)

RussTedrake, 2018-08-27:
the primary motivation for using sdformat is that if (when?) we want to add an additional tag, we solicit upstream support for it.

but I am ok with doing our own parsers if the confidence is not there.  as just discussed in the drake developers meeting, a number of folks will not switch from RBP => MBP until we have urdf support there.

RussTedrake, 2018-08-30:
Just to document it here, the workflow we have been using has been to use the gazebo tools for urdf=>sdf conversion, with some manual checking/fix-ups.  They seem to be different/better than the sdformat version.