RobotLocomotion/drake #9286: multibodyplant documentation should mention parsing

RussTedrake, 2018-08-21:
The documentation here: https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1multibody__plant_1_1_multibody_plant.html#details 
is great, but is not the way 99.5% of users will interact with the code.  Please add the use case of loading from a urdf/sdf here?

amcastro-tri, 2018-08-22:
I wonder if we'd like something more in the form of a tutorial for this. In that case, would it be part of the Doxygen or should we somehow reference a pdf? ideas or preferences?

RussTedrake, 2018-08-22:
I think a lot of projects write their tutorials in sphinx, too.  It’s worth considering.

Even a short note in the doxygen could go a long way, too.

jwnimmer-tri, 2018-11-19:
With the instructions merged in #9312 and my amendments pending in #10058, is there anything remaining to do here?

RussTedrake, 2018-11-20:
Lgtm

closed 2018-11-20 by RussTedrake