RobotLocomotion/drake #9290: Git clone instruction for drake is not correct

DamrongGuoy, 2018-08-21:
On drake [web page for source installation](http://drake.mit.edu/from_source.html) the command
`git clone git@github.com:RobotLocomotion/drake.git`
should have been
`git clone https://github.com/RobotLocomotion/drake`
using URL to the repo.

I'll fix it as an exercise for my first pull request.

SeanCurtis-TRI, 2018-08-21:
There's not an error in the documentation -- all of my clones are via the `git` protocol. It may be there is a *different* error. But moving to `https` isn't necessarily the solution.

soonho-tri, 2018-08-21:
FYI, in the doc, we have this precondition:

> The following instructions assume you have uploaded your public SSH key to your Github account.

I guess it was not the case and caused the error. Could you check?

DamrongGuoy, 2018-08-21:
Thank you, @soonho-tri .  Indeed I did not upload my public SSH key.  This is the error I got.
```
$ git clone git@github.com:RobotLocomotion/drake.git
Cloning into 'drake'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
I will upload my public SSH key and confirm again.  On the other hand, moving to `https` yesterday let me clone and build drake successfully.

SeanCurtis-TRI, 2018-08-21:
Your life will be much easier once you get the SSH keys uploaded. Pushing and pulling are *greatly* streamlined.  :)

DamrongGuoy, 2018-08-21:
Thank you, @SeanCurtis-TRI.  I agree and close the issue now.

closed 2018-08-21 by DamrongGuoy