RobotLocomotion/drake #9309: Typos/mistakes in the equations in publication Tedrake et al., 2015

hwiedDFKI, 2018-08-27:
## Description
I read your paper "A closed-form solution for real-time ZMP gait generation and feedback stabilization" from 2015. Unless I'm mistaken there are some errors in the equations (9ff. and in Algorithm 2 (<i>a<sub>j</sub></i>)) regarding the matrix calculations, as the dimensions don't match. Do you have a reviewed version of this paper with correct equations?

I suppose in eq. 9 you meant <i>Q<sub>1</sub> = C Q C<sup>T</sup></i> and in the following equations <i>Q<sub>1</sub></i> is meant instead of <i>Q</i>.
But in algorithm 2 I'm confused how it's to be meant, as e.g. for <i>a<sub>j</sub></i>, <i>x</i> and <i>b<sub>j,1</sub></i> have different dimensions.
Thank you very much in advance :)

hwiedDFKI, 2018-08-27:
Or could you give me a hint where these algorithms are implemented in detail.

hongkai-dai, 2018-08-27:
I think equation 9 is correct, it should be Q₁ = CᵀQC.

`C` is of size `n_y * n_x`, `Q` is of dimension `n_y * n_y`, as it is the cost matrix for `y` in the objective function (4), so CᵀQC gives you a positive semidefinite matrix of dimension `n_x * n_x`.

The algorithm is implemented in `systems/controllers/zmp_planner.h/cc`

hwiedDFKI, 2018-08-28:
Thank you very much. I missed that Q was the weight matrix for the outputs and you transformed it to the states in equation 9. 

Also the code helped me to understand how you meant algorithm 2. In the paper it was challenging for me to see where you use beta, and where you take only the top half of b.

Thank you very much. And by the way, thanks that your code is so well documented, with reference to the equations, comments and fitting variable names. That's quite rare.
Best wishes

closed 2018-08-28 by hwiedDFKI