RobotLocomotion/drake #9310: Drake build now requires SNOPT_PATH setting

sherm1, 2018-08-27:
A recent update to Drake's build system (not sure which one) apparently introduced a requirement to set SNOPT_PATH, even for those of us who don't have snopt. I encountered this error:
```
[0 / 2] BazelWorkspaceStatusAction stable-status.txt
ERROR: /home/sherm/.cache/bazel/_bazel_sherm/e6c506c79a0a2d8ce8be77e2bf55a048/external/snopt/BUILD:9:1: Couldn't build file external/snopt/error-message.h: Executing genrule @snopt//:error-message failed (Exit 1).
ERROR: Repository rule @snopt failed: SNOPT_PATH of '' is malformed
```

@soonho-tri suggested setting `SNOPT_PATH=git` and that allowed me to build.

[Here](https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1534896766000100) is the slack discussion.

sherm1, 2018-08-27:
@jwnimmer-tri's comment from slack:
> Jumping back to the CLion + SNOPT=git question above -- is there an open Drake issue about this?  @sherm1 @soonho-tri I didn't immediately see one.  At minimum, we should update `clion.rst` and/or `bazel.rst` with any required solution.  However, I also suspect that the right answer is a per-user bazelrc file, not /etc/environment.  (This is how Anzu does it.)

jwnimmer-tri, 2018-08-27:
Are you sure that you have SNOPT disabled?  That error is only supposed to be shown if you ask to compile with SNOPT enabled, but did not provide a path.  What build target are you running?  What are the `--config` flags?

jwnimmer-tri, 2018-08-27:
(Check your CLion project settings, or `$HOME/.bazelrc` or any other rc files you might have set up?)

sherm1, 2018-08-27:
I'm not sure -- should I have actively disabled something?

jwnimmer-tri, 2018-08-27:
No, by default it should be off.  However, in the mists of time maybe you set `--config snopt` at some point and never turned it off again?

sherm1, 2018-08-27:
Oh, actually my `.bazelrc` looks like this:
```
build --config snopt
test --test_tag_filters=-gurobi,-mosek
```
That does look suspicious! I have no idea where that came from -- likely I copied it mindlessly from someone else. If I remove the first line, that should remove the environment variable requirement I presume?

jwnimmer-tri, 2018-08-27:
You should remove (or comment out) _both_ lines, but after that -- yes, the intent is that SNOPT_PATH can (and should) be unset.

sherm1, 2018-08-27:
OK, will do and test. If that's fine then I don't think there is any issue here and I'll close it.

jwnimmer-tri, 2018-08-27:
Would it have helped if the error message said something more like this?
```console
ERROR: Repository rule @snopt failed: SNOPT was enabled with `--config snopt` or
`--config everything` but SNOPT_PATH of '' is malformed.
```

sherm1, 2018-08-27:
Yes, but even more so if it had said 
```
ERROR: Repository rule @snopt failed: SNOPT was enabled with `--config snopt` or
`--config everything` (possibly in your `.bazelrc`) but SNOPT_PATH of '' is malformed.
```

Without the mention of `.bazelrc` I would have stumbled around for some time because I didn't even know that was a thing. (Probably did for 5 minutes long ago when someone gave me one to copy but since then it drifted out of my consciousness.)

sherm1, 2018-08-27:
I commented out my `.bazelrc` and got rid of my `SNOPT_PATH` environment variable. I'm not seeing any build problem now.

I'll leave this open in case you want to improve the error message. Otherwise, feel free to close.

PR #9449 (2018-09-13): snopt: Improve error message when SNOPT_PATH is unset
Closes #9310.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9449)
closed 2018-09-13 by jwnimmer-tri