RobotLocomotion/drake #9311: MultibodyTree + pydrake: Binding feature requests

EricCousineau-TRI, 2018-08-27:
Suggested order of operations:

- [ ] Load an SDF and query basic topology from `MultibodyPlant`
- [ ] Simulate a `MultibodyPlant`
- [ ] Visualize a `MultibodyPlant`
- [ ] Compute kinematics for `MultibodyTree` / `MultibodyPlant`
- [ ] Compute dynamics for `MultibodyTree` / `MultibodyPlant`
- [ ] Construct a `MultibodyPlant` / `MultibodyTree` programmatically

\cc @amcastro-tri

pangtao22, 2018-08-31:
In Pset1 of 6.881 (to be released on Sep 11), we are using the following RBT functions to control the robot and in testing:
```
rbt.B
rbt.doKinematics
rbt.CalcBodyPoseInWorldFrame
rbt.inverseDynamics
```

Not used in pset1 but could be used for future psets:
```
rbt.relativeTwist
rbt.geometricJacobian
rbt.geometricJacobianDotTimesV
```

Thanks!

EricCousineau-TRI, 2018-08-31:
Per f2f with @amcastro-tri, for the actuation matrix / projection matrix `B`, we may need some sugar methods to obtain the projection matrices.
@hongkai-dai Can I ask if you'll be submitting a PR for getting things such as the state and Jacobians in a specified order? If so, I can then build on top of that PR to get `B`.

hongkai-dai, 2018-08-31:
@EricCousineau-TRI I do not think I will send a PR on getting things like state and Jacobians. By discussing with @amcastro-tri , I will PR a method to get the bounds on the generalized positions.

tri-ltyyu, 2018-11-05:
Check with @EricCousineau-TRI on status

EricCousineau-TRI, 2018-11-05:
Updated checkboxes; we can consider shifting the remaining bits that are not immediately necessary to separate issues.

EricCousineau-TRI, 2020-02-14:
AFAIK, with contributions from @gizatt, @mpetersen94, and other Drake authors, we've generally covered this list. Closing for now.

closed 2020-02-14 by EricCousineau-TRI