RobotLocomotion/drake #9347: CLion on macOS no longer works

jwnimmer-tri, 2018-08-31:
Discussed at length in slack; https://drakedevelopers.slack.com/messages/C36L0D7D4/convo/C36L0D7D4-1535410503.000100/.  Exploratory testing looks like #8881 is what broke it.  The default macOS toolchain is thought to not be a C toolchain in the IDE.

jwnimmer-tri, 2018-08-31:
See also bazelbuild/intellij#109 and bazelbuild/intellij#397.

jwnimmer-tri, 2018-08-31:
\CC @RussTedrake

jwnimmer-tri, 2018-08-31:
The commit on https://github.com/jwnimmer-tri/drake/commits/macos-clion-workaround appears to work around the problem for now.  Users have to merge it into their local copy, but not commit it to master.

RussTedrake, 2018-09-01:
Here's the one-liner 
```
git fetch https://github.com/jwnimmer-tri/drake macos-clion-workaround && git cherry-pick daca23091e40bde66bd89f8f206894513fdc7ed6
```

jwnimmer-tri, 2018-09-06:
It is possible that, instead of the above patch, setting `BAZEL_USE_CPP_ONLY_TOOLCHAIN=1` in the environment (before launching CLion) would also get things working on macOS.

RussTedrake, 2018-09-07:
Tested the environment variable patch, and confirmed that it works!  Woohoo!

PR #9448 (2018-09-14): clion: Add more and consolidate macOS tips
Closes #9347.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9448)
closed 2018-09-14 by jwnimmer-tri