RobotLocomotion/drake #9353: Should structurally equal uninterprted_functions compare as formula-equal?

jwnimmer-tri, 2018-09-01:
This (newly-proposed) test case passes.  I'm not sure that it should:
```
TEST_F(SymbolicFormulaTest, EqualTo0) {
  const Expression expr1 = uninterpreted_function("func", {var_x_});
  const Expression expr2 = uninterpreted_function("func", {var_x_});
  EXPECT_EQ((expr1 == expr2).to_string(), "True");
}
```

Many built-in simplification rules for arithmetical Expression operations (like subtraction, addition factory, etc.) assume that if their terms are structurally equal, that we may cancel or gather terms.  I am not sure this should be the case for uninterpreted functions.

I thought of this bug in the context of #9338.  There, we have something akin to this:
```
Variable m00, m01, m02, m10, m11, m12, m20, m21, m22 = ...
Matrix3<Expression> input =  m00, m01, m02, m10, m11, m12, m20, m21, m22;
Matrix3<Expression> normalized = ProjectToOrthonormal(input);
```
For doubles, the ProjectToOrthonormal would do SVD.  Since that isn't realistic for symbolic variables, I proposed a symbolic `ProjectToOrthonormal` implementation specialized for Expression which does this:
```
Matrix3<Expression> result;
Variables unbound = input(0, 0).GetVariables() + input(0, 1).GetVariables() + ...;
result(0, 0) = uninterpreted_function("orthonormal_projection", unbound);
result(0, 1) = uninterpreted_function("orthonormal_projection", unbound);
result(0, 2) = uninterpreted_function("orthonormal_projection", unbound);
...
```
My intent was to label each element of the result matrix as "some unspecified function based on these variables".  However, if the user were to compute `result(0, 0) - result(0, 1)`, they would get zero, or `result(0, 0) == result(0, 1)` would be `True`.  Neither is correct in this case.

I guess one could argue that I should have each element of the result be some _different_ unspecified, function, as in:
```
...
result(0, 0) = uninterpreted_function("orthonormal_projection00", unbound);
result(0, 1) = uninterpreted_function("orthonormal_projection01", unbound);
result(0, 2) = uninterpreted_function("orthonormal_projection02", unbound);
...
```
and while that would disambiguate two terms from the same result, comparing `resultA(0, 0) == resultB(0, 0)` still might be `True` (if the variables were the same set), since they are both function `"orthonormal_projection00"`.

So in short, I think that perhaps we need to fix symbolic logic, to treat this term as different that the rest.  If its _only_ mean to imply sparsity, they it can't really be canceled like other terms.

Alternatively, if it were an `uninterpreted_function` over an _Expression_ (or more likely, tuple of _Expressions_) instead of merely over a _Variables_, then maybe it would be good enough?  I could pass the individual input elements into the uninterpreted function (along with an index salt, for example) and then structural equality _would_ imply cancel-ability.

@soonho-tri any thoughts?

sherm1, 2018-09-01:
My 2 cents: I think
```c++
TEST_F(SymbolicFormulaTest, EqualTo0) {
  const Expression expr1 = uninterpreted_function("func", {var_x_});
  const Expression expr2 = uninterpreted_function("func", {var_x_});
  EXPECT_EQ((expr1 == expr2).to_string(), "True");
}
```
_should_ return true since the function and its argument are identical.

In the case of the orthonormal project, you shouldn't have used a per-element function. Instead a vector-valued function `orthonormal_projection(R)` would be the correct symbolic expression -- there is no way to do this elementwise that would be meaningful.

jwnimmer-tri, 2018-09-01:
Yup, that's essentially what my final paragraph proposes.  We need to track the entire `R` as a whole (not just `R`'s free variables).

That is, the argument to an `uninterpreted_function` should be a `std::vector<Expression>`, not a `std::set<Variable>`.

Then we would have (for input R of type `Matrix3<Expresion>`) ...
```
Expression r00 = R(0, 0);
Expression r01 = R(0, 1);
Expression r02 = R(0, 2);
Expression r10 = R(1, 0);
Expression r11 = R(1, 1);
Expression r12 = R(1, 2);
Expression r20 = R(2, 0);
Expression r21 = R(2, 1);
Expression r22 = R(2, 2);
result(0, 0) = uninterpreted_function("ortho_proj00", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(0, 1) = uninterpreted_function("ortho_proj01", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(0, 2) = uninterpreted_function("ortho_proj02", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(1, 0) = uninterpreted_function("ortho_proj10", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(1, 1) = uninterpreted_function("ortho_proj11", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(1, 2) = uninterpreted_function("ortho_proj12", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(2, 0) = uninterpreted_function("ortho_proj20", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(2, 1) = uninterpreted_function("ortho_proj21", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(2, 2) = uninterpreted_function("ortho_proj22", make_vector(r00, r01, r02, r10, r11, r12, r20, r21, r22))
```

... written out longhand, because our `Scalar` type is never vector- or matrix-valued; it's always a double or Expression or etc.

jwnimmer-tri, 2018-09-01:
One could also cheat the index constant into the argument list, and reuse the function name:
```
result(0, 0) = uninterpreted_function("ortho_proj", make_vector(0.0, r00, r01, r02, r10, r11, r12, r20, r21, r22))
result(0, 1) = uninterpreted_function("ortho_proj", make_vector(1.0, r00, r01, r02, r10, r11, r12, r20, r21, r22))
...
...
```

sherm1, 2018-09-01:
I don't like either of those -- trying to do this elementwise is a losing battle. What I meant was that the function itself should return a matrix and take a matrix argument. Something like
```
R = uninterpreted_matrix_function("orthonormal_projection", matrix);
```

jwnimmer-tri, 2018-09-01:
What _specifically_ is the problem with doing it elementwise?  At the end of the day, a `T` is a single scalar, so we have to map everything into that terminology eventually.  We could sugar it up by embedding row and column loops inside a Matrix-like helper, but under the hood its still going to be all scalars.

sherm1, 2018-09-01:
The problem is that there is no meaningful relationship between the elements of the source matrix and the elements of the orthogonalized matrix. _Each_ element in the result comes from matrix operations that involve _every_ element of the source. But that's not 9 separate operations -- it is a single matrix operation on the source yielding a new matrix, whose elements are the result. Expressing the operation elementwise loses the beautiful semantics of the matrix math.

jwnimmer-tri, 2018-09-02:
While I can see the appeal of having our fundamental datatype in Drake be matrix-valued instead of scalar-valued, that's not what Drake has chosen to implement.  I don't see preserving "the beautiful semantics" as an interesting requirement -- at the end of the day, everything in Drake boils down to a Scalar, so whatever we do has to map onto a scalar somehow.  In my most recent proposal above, we have preserved (1) the set of variables that feedthrough the computation and (2) uniqueness such that an identical output is produced if and only if the inputs are identical.  That seems good enough to me?  A sufficiently smart user of the symbolic form's result could even match the `"orthonormal_projection"` function name, and substitute it for their own projection primitive.

In any case, in this issue I am not advocating handling orthonormal projection symbolically.  I am saying that `uninterpreted_function` is dangerous should not be used in its current form.

Using a different example would probably motivate that discussion better.  How about this function:
```c++
/// Returns a sorted copy of the given vector.
template <typename T>
Eigen::VectorX<T> Sorted(const Eigen::VectorX<T>& input) {
  // .. default implementation uses std::sort or similar ...
}
```

That default implementation would suit for most Scalars.  However, for Expression, if there are unbound variables in the input, the `if-then` control flow in most sorting algorithms would fail, so we would likely need to overload this for Expression:
```c++
// Specialized for Expression.
template <>
Eigen::VectorX<symbolic::Expression> Sorted(const Eigen::VectorX<symbolic::Expression>& input) {
  // ???
}
```

What should we write for `???`?

We _could_ actually write a completely symbolic sorting routine -- for example, using bubble sort with the maximum number of iterations (N = `input.size()`) and using `min()` and `max()` to form the `swap()` primitive.  However, that could lead to extremely large expression tree as the result (`O(N * 2^N)`), and is unlikely to be helpful for most consumers of the symbolic form.

Instead, I would recommend an `uninterpreted_function`:
```c++
// Replace the current symbolic_expression.h signature with this one:
Expression uninterpreted_function(std::string name, std::vector<Expression> args);

// Private helper for Sorted().
namespace {
Expression ith_smallest(int i, const Eigen::VectorX<Expression>& input) {
  const int N = input.size();
  std::vector<Expression> args;
  args.resize(N + 1);
  args(0) = i;
  for (int i = 0; i < N; ++i) {
    args(i + 1) = input(i);
  }
  return uninterpreted_function("ith_smallest", std::move(args));
}
}  // namespace

// Specialized for Expression.
template <>
Eigen::VectorX<Expression> Sorted(const Eigen::VectorX<Expression>& input) {
  const int N = input.size();
  Eigen::VectorX<Expression> result(N);
  for (int i = 0; i < N; ++i) {
    result(i) = ith_smallest(i, input);
  }
  return result;
}
```

I believe this captures the operation relatively succinctly, and demonstrates why `uninterpreted_function` _must_ take an Expression list, not a Variable set, as an argument.  If the input was, e.g., `[2 * x, 2 * y, 2 * z]` then the `ith_smallest(0, ...)` should _not_ be `min(x, y, z)` but rather _must_ take into account the factor of two.  Similarly for opportunistic term cancellation / equality / etc.

sherm1, 2018-09-03:
> While I can see the appeal of having our fundamental datatype in Drake be matrix-valued instead of scalar-valued, that's not what Drake has chosen to implement. 

If you want a symbolic math package to deal with _unevaluated_ matrices, it has to support matrices directly. They are more than a collection of their (unknown) elements. For example, say you want to support orthogonalization of a symbolic 3x3 matrix M whose  variable values are currently unknown. You can defer the result by producing the expression `orthogonalize(M)`. If you now encounter `orthgonalize(M)*transpose(orthogonalize(M))` (common in rotation sequences) you should simplify that to a 3x3 identity matrix. There is no elementwise representation of this concept. Factorizations, determinants, eigenvalues, positive definiteness, etc. are similar concepts. You can always perform _operations_ elementwise but can't represent matrix _concepts_ that way since they are holistic properties of the collection of elements.

If our symbolic capability doesn't yet include matrices, we should not attempt to cobble matrix concepts together elementwise. Instead, let's not support those now and enhance our symbolic capability when we actually have a need to do so. I don't think we currently have such a need? 

Regarding the sorting example, I don't claim any particular insight. I would prefer just `sorted(args)` to spelling it out elementwise, since that smells like extra computation later. Why do you see the elementwise expansion as preferable?

jwnimmer-tri, 2018-09-03:
I have absolutely no idea what it could possibly even look like to "support matrices directly".  You will have to write up a sample C++ signature or two, for me to be able to follow along.  Drake uses `Eigen::MatrixBase<Derived, ...>` etc where `Derived` has to be _some_ scalar.  Unless we want to stop using Eigen, we have to individualize the scalars that make up our matrices.

I still disagree that you can't do it elementwise.  I could easily write reductions that perform the orthog-transpose-orthog => identity symbolically, using just symbolic scalars without some overarching matrix concept.

In any case, I still don't think that's relevant to this issue, though.  This issue is that `uninterpreted_function` is broken.

I think my `Sorted()` example didn't help.  It was too complicated.

How about we consider just `Factorial(e)`?  Conceivably we could add a special-case primitive Expression term, but let's say we wanted to do it with `uninterpreted_function`.

I'd posit that taking the factorial of `2 * x` should not be `uninterpreted_function("factorial", {x})` but rather `uninterpreted_function("factorial", [2 * x])`.  The fact that we can currently only do the former is a defect.

sherm1, 2018-09-03:
Oh, I totally agree we need expressions for the arguments.

Eigen is only capable of representing values of matrices and operating on them. That is not a problem. If I understood your proposal, you were looking to represent the concept of an orthogonalized matrix, without actually performing the orthogonalization computation. There is no Eigen equivalent.

soonho-tri, 2018-09-04:
Sorry for being late.

1) I agree that we need to change the current design of `uninterpreted_function` so that it will take a `std::vector<Expression>` instead of a set of `Variables`. I'll open a PR soon.

2) For now, all symbolic expressions are typed as Scalar. I think what @sherm1 is describing is another type (that we do not have) in symbolic language - a matrix type so that matrix-level operations can be written without unfolding them to the scalar level.

sherm1, 2018-09-04:
> I think what @sherm1 is describing is another type (that we do not have) in symbolic language - a matrix type so that matrix-level operations can be written without unfolding them to the scalar level.

Right. I'm not saying we should do that now; I'm saying that scalarizing is not the appropriate way to handle matrix-level operations so we shouldn't do that. If it becomes important, we can enhance the symbolic code to have a matrix type.

PR #9384 (2018-09-12): Symbolic uninterpreted_function takes a vector of expressions instead of a set of variables
Close #9353 

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9384)
closed 2018-09-12 by jwnimmer-tri