RobotLocomotion/drake #9358: Clarify the joint type for a newly added rigid body in the documentation

hongkai-dai, 2018-09-02:
I would like to add a floating body to a `MultibodyTree`. According to slack discussion https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1535829592000100

eric.cousineau [12:19 PM]
When working in Python, looking at Sammy's code, and talking with @alejandro, it looks like you just add the body, and if you do not weld it, then it is implicitly a floating joint.

But this is not clear from `AddRigidBody` method in `MultibodyTree`. We should clarify it in the documentation.

amcastro-tri, 2018-09-04:
any suggestions on where this doc should leave? 
also, any volunteers to write this down?

hongkai-dai, 2018-09-04:
I would suggest to add this doc in `AddRigidBody`.

closed 2024-03-20 by jwnimmer-tri