RobotLocomotion/drake #9359: on-demand build instructions are stale?

RussTedrake, 2018-09-02:
I just tried to follow the instructions here:
http://drake.mit.edu/jenkins.html#running-an-on-demand-build
but failed.

instead i clicked on the experimental build that I wanted... could not see anywhere I could click "Build with parameters", but did find the text "@drake-jenkins-bot mac-highsierra-clang-bazel-experimental please" and was able to run that from the github console... which was ideal for what I was looking for this time.

did I miss the "Build with parameters" button (in which case we could make the instructions better)?  or can/should we update the instructions to suggest what I eventually stumbled into?

jwnimmer-tri, 2018-09-02:
The "Build with parameters" is only shown if you are logged in.  In the upper right, did it offer "Log in" button or (which means you are not logged in) or "Log out" (which means you are already logged in).  When logged in, I do see "Build with parametesrs".

RussTedrake, 2018-09-02:
I see.  That resolved my issue.  Thanks.  
The instructions do say "Log in", but the link below seemed to have worked (it was not clear to me that I had not completed logging in when I was searching for the gui element).  Perhaps a minor tweak to the instructions would save others time in the future (or me, when I try again in 2 months) ?

PR #10292 (2018-12-21): doc: Clarify the jenkins login step
Closes #9359.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10292)
closed 2018-12-21 by jwnimmer-tri