RobotLocomotion/drake #9367: multibody: Provide ease-of-use mechanism to communicate state / input / output w/ RBT

EricCousineau-TRI, 2018-09-03:
At present, @pangtao22 would like to be able to use MBP for doing dynamics and kinematics, but would like to leverage existing components which were designed for RBT for things like custom visualization (e.g. [`MeshcatVisualizer`](https://github.com/RussTedrake/underactuated/blob/a1e5a8adf7c00f1b4a17045cbbf482c9a37b8fe4/src/underactuated/meshcat_rigid_body_visualizer.py)).

It does not appear as there's anything that can readily provide information on a joint's positions within the state vector.

\cc @hongkai-dai @amcastro-tri

SeanCurtis-TRI, 2018-09-04:
Also worth noting that the idea of a general visualization interface on SG is on an official shopping list (see #8473 -- the item "Support visualization blocks as a consumer of GeometrySystem").

Ultimately, SG will offer a couple of ports: scene_configuration and scene_state which should, ultimately, allow connection to any system that wants to visualize the world according to its details.

EricCousineau-TRI, 2018-12-08:
Closing, as this is no longer relevant.

closed 2018-12-08 by EricCousineau-TRI