RobotLocomotion/drake #9370: Fail fast when `install_data()` rule is left dangling (not depended-on by `//:install)

jwnimmer-tri, 2018-09-03:
Installed example models need to be called out at https://github.com/RobotLocomotion/drake/blob/dc6232f720ca75b27b1d920b91108c7ad27d7f04/examples/BUILD.bazel#L39.  This is easy to forget.

We should add a linter check (in `bazel test //examples/mypkg/...`) that its install rule is cited.  Now that snopt is `genquery()`-able, I think this is easy.

Probably _all_ install rules should have the linter, not just examples nor install_data.

As noted in slack a while ago https://drakedevelopers.slack.com/messages/C2PMBJVAN/convo/C2PMBJVAN-1523271986.000134/.

closed 2018-09-07 by jwnimmer-tri

jwnimmer-tri, 2018-09-13:
For the record: the original fix had to be disabled, but #9446 fixes this for good.

soonho-tri, 2018-09-13:
I think if one runs `bazel run //:install --`, it will build Drake and install it *without* running the `install_lint`. Can we change it to enforce running `install_lint`?

jwnimmer-tri, 2018-09-13:
We _could_ turn this lint error into a build failure instead of test failure (though it would involve some hijkins), but I'm not sure that we'd want to?  For anyone installing from `master`, CI would already ensure that the install rules are lint-free.  For anyone installing from local edits, well, if you're not running the tests on the package you're modifying, I think you get to keep the pieces.

soonho-tri, 2018-09-13:
I see. I think my lesson is to make sure adding the Mac CI builds to the PRs which changes the install logic.

PR #9395 (2018-09-07): Add linting for dangling install rules
Closes #9370.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9395)