RobotLocomotion/drake #9391: Weld joint workflow in MultibodyPlant is cumbersome and inefficient

RussTedrake, 2018-09-06:
Compared to the RigidBody* equivalent, where the AddRobotFromSdf workflow takes a pose (and optional parent), the workflow in MBP/SceneGraph requires many more lines to achieve the same goal (by adding a WeldJoint).

I suspect this workflow also has consequences on the implementation / performance.  Even if I add static geometry into the scene via the parsers -- by adding from sdf then adding a weld joint -- it appears to not be registered as AnchoredGeometry with SceneGraph.  cc @SeanCurtis-TRI 

(Note, however, that if we do achieve that goal, then we need to test the situation where the same sdf is added multiple times into the environment.  Right now all of the links will have identical names, and it is only the model_instance() that keeps them distinct).

Thoughts?

RussTedrake, 2018-09-06:
I'm also curious if @amcastro-tri and @SeanCurtis-TRI would recommend parsing sdf files that contain only static (anchored) geometry directly into SceneGraph (never tell MBP about it, but make it available for collision queries, etc).

SeanCurtis-TRI, 2018-09-06:
I really need to write up a design doc for what I have in mind in parsing....

The short answer is, "Yes. What you said."

In my ideal world, MBP only parses links and, if a `SceneGraph` is provided, registers a frame for each body/link. It then produces a map between some unique body identifier from the sdf (not 100% sure what that would be, but name-related string will probably be it) and the registered frame id. It never knows about or looks at geometry. Then `SceneGraph` takes the map, the `SDF` object (and possibly a collection of additional parameters), and passes through the SDF again, registering the geometry itself. 

For optimization, the map could include additional information (such as the link with the given name is effectively welded to the world -- leading to its geometry being registered as anchored).

The goal is for MBP (and any other geometry source) to only know about the things it has to know about.

amcastro-tri, 2018-09-06:
Optimization to anchored geometries could live in a different issue IMO.

In the short term, @RussTedrake you are suggesting to modify the API of `AddModelFromSdfFile()` so that it takes a frame and the pose of the new model in that frame?
That seems good to me, just double-checking that's what you'd want.

RussTedrake, 2018-09-06:
Yes.  Taking the frame and the pose is the short term ask.

Also — from slack — wondering if the way that the kuka model avoids being a floating base can be handled by this.  I think now that the reason for the current state of the sdf is for gazebo compatibilty.

Cc @EricCousineau-TRI

RussTedrake, 2018-09-07:
@EricCousineau-TRI 
+1 to using SDF worlds.  That would go a long way.

Regarding API -- i was thinking that it would be optional arguments (for pose and frame) at the end of the existing entry point.  But being clear is as important as being concise.  Your proposal above seems reasonable.

I think that our current `AddModelFromSdfFile` is morally equivalent to the [include](http://sdformat.org/spec?ver=1.6&elem=model#model_include) tag in sdformat.  I *think* that if the included file contains a joint connecting it to the magical "world" link, then this effectively adds the weld joint.  But even though the included file specifies a joint to the world at a particular pose, the include statement can also [add a pose element](http://sdformat.org/spec?ver=1.6&elem=model#include_pose) to put one more transform into the tree before the inclusion.  It is this include pose that I think we're missing in our current API.

amcastro-tri, 2018-09-10:
working on this right now

amcastro-tri, 2018-09-10:
SDF does not encode the concept of a "tree". So in principle, there is nothing that mark any link in particular as being the "root" of that model. 
That is, I believe that something like:
```c++
model_instance = WeldToWorld(AddModelFromSdfFile(...), X_WM);
```
would just not work because in principle (even if after finalization and **internally** that does happen),  there is nothing that marks a particular body as being the root of the model instance.
That is why we have the current workflow.

However, there clearly is room for improvement in terms of additional sugar. What would you guys think about something like:
```c++
robot = AddModelFromSdfFile(&plant, ...);
gripper = AddModelFromSdfFile(&plant, ...);

// Weld gripper frame G to end effector frame E with pose X_EG
plant.WeldFrames(
    // Frame E.
    plant.GetFrameByName("end_effector", robot), 
    // Frame G.
    plant.GetBodyByName("main_body", gripper).body_frame(), 
    // Pose of gripper frame G in the end effector frame E.
    X_EG); 

```

RussTedrake, 2018-09-11:
The workflow that is more painful right now is not welding the gripper to the arm, but the way that the arm is welded to the world.

In iiwa14_no_collision.sdf, there is 
```
    <joint name='weld_base_to_world' type='fixed' drake_ignore='true'>
      <child>iiwa_link_0</child>
      <parent>world</parent>
    </joint>
```
The `drake-ignore` is misleading -- that's only for the RBT parser.  

The current MBT parser treats this as a weld joint between the arm and the world frame.  So you cannot place the arm into any location except 0,0,0.  We've had to make local edits to the sdf file in order to move it.

amcastro-tri, 2018-09-11:
With the API proposed above, fixing the the arm to the world would look like:
```c++
plant.WeldFrames(
    // World Frame.
    plant.world_frame() 
    // Rotob base Frame B.
    plant.GetBodyByName("base", robot).body_frame(), 
    // Pose of the base frame B in the world frame W.
    X_WB); 
```

would this solve the problem?

RussTedrake, 2018-09-11:
It's not enough.  The iiwa sdf that is currently in master comes out of the parser _already welded_ to the world frame at 0,0,0.  We have had to make local edits to it to remove that joint.  But I think the sdf is the way that it is for compatibility with gazebo.

amcastro-tri, 2018-09-11:
@sammy-tri created this file. I believe we welded the base within the SDF since MBPs parser would otherwise create a free floating arm (in accordance to the SDF specification). 
@sammy-tri, is that right? could we just remove the `weld_base_to_world` from the SDF and start using the API proposed above?

sammy-tri, 2018-09-11:
I think I added the `weld_base_to_world` joint for compatibility with MultibodyPlant as a hack, not for compatibility with gazebo.  If we have a nicer API (like above) I think that joint can/should be removed (as well as the comment at the top of the SDF discussing it).

RussTedrake, 2018-09-11:
thanks!
cc @pangtao22

amcastro-tri, 2018-09-11:
#9434 in progress.

amcastro-tri, 2018-09-12:
I believe the merging of #9434 closes this? yes/no?

RussTedrake, 2018-10-03:
@amcastro-tri -- I still think there is a deficit in the current workflow.  I'm using the WeldFrames() api.  But @EricCousineau-TRI also mentioned a concept of `FloatInWorld(model_instance, pose)` that is missing.  

For instance, I'm building a system that inserts a table, then inserts a bunch of objects that I'd like to start on the table.  The objects should of course be floating joints, but there is no way (I think) of setting the coordinate system for the objects to be relative to the table.  So somehow all of my state initialization logic downstream has to remember the coordinate system of the table.  I really want the ability to add an additional pose in the tree at the top of a model_instance that transforms the frame of that floating subtree.  It could be in the sdf parser call, or immediately after, but I think right now it's simply missing.

amcastro-tri, 2018-10-05:
I see the problem now. I'll see how to address it.

amcastro-tri, 2018-10-17:
- Welding frames implemented in #9434 
- Ability to set a free body pose in an arbitrary anchored frame implemented in #9664.

closed 2018-10-17 by amcastro-tri