RobotLocomotion/drake #9398: pybind: Figure out why `py_reference` is necessary for a function return a pointer

EricCousineau-TRI, 2018-09-06:
See this comment: https://github.com/RobotLocomotion/drake/pull/9392#issuecomment-419181776

EricCousineau-TRI, 2019-02-09:
Just got bit by this again in Anzu, when tinkering with `FixedInputPortValue::GetMutableData`.

Going to move up in my work queue, may raise in priority, as I've lost a bit of time to this.

EricCousineau-TRI, 2019-02-13:
@avalenzu may have ran into some form of this...

m-chaturvedi, 2019-04-28:
> See this comment: #9392 (comment)

So, it looks like when the return value is a pointer, [`automatic` resolves to `take_ownership`, not `reference`](https://pybind11.readthedocs.io/en/stable/advanced/functions.html).  If we change `reference` to `automatic_reference` the test passes.

>`return_value_policy::automatic` [...] This policy falls back to the policy `return_value_policy::take_ownership` when the return value is a pointer. Otherwise, it uses `return_value_policy::move` or `return_value_policy::copy` for rvalue and lvalue references, respectively. See above for a description of what all of these different policies do.

\cc @jamiesnape

EricCousineau-TRI, 2019-04-28:
D'oh! That would make sense lol.
I guess I was always thinking that `automatic_reference` was the way to go...

The bad part of that, though, is that I think that I have let that wrong assumption propagate through the code base :( There are probably now many places that need `py_reference`.

I will try to think of a way to suss those out with a temp `pybind11` patch.

EricCousineau-TRI, 2019-04-28:
Here's the temp patch:
https://github.com/pybind/pybind11/commit/87f74e94de3059ac47ad3d14335584762927c7f1

Here's some WIP work to fix the issues:
https://github.com/EricCousineau-TRI/drake/commit/83e3a8554462cac5401b52f8d2d1b156b044cbb9

One issue is, I don't know why `pybind11` didn't consistently throw errors before. To me, it seems like there are two modalities:
* With `cpp_function` returns, `T*` somehow ends up resolving to just returning `rvp::reference`. This may happen in two ways:
  * It resolves `rvp::automatic => rvp::take_ownership`; ownership cannot be taken for `T*`, so it simply returns the reference silently.
  * It resolves `rvp::automatic_reference => rvp::reference`, and things are OK.
* With `py::cast(...)`, it seems that `T*` somehow always resolves to a copy, which is why those TODOs are there.

@m-chaturvedi When you have a chance, can you investigate why `pybind11` has a difference between those two? Hopefully it's documented in RTD like you find...

m-chaturvedi, 2019-04-29:
> But I still don't know why py::cast() does something different...

Firstly, [this `cast`](https://github.com/RobotLocomotion/pybind11/blame/drake/include/pybind11/cast.h#L1974) is actually what is resolved, [not that one](https://github.com/RobotLocomotion/pybind11/blame/drake/include/pybind11/cast.h#L1947) when you don't pass any `policy`.  
The resolution happens because `T&&` is preferred over `const T&`, as pointed [out here](https://stackoverflow.com/questions/22167235/understanding-template-argument-deduction-with-rvalue-lvalue).
When we add `reference` as the `return_value_policy`, the other `cast`, that you point to, is resolved.

Secondly, the segfault is happening because of the `move` you have have, [here](https://github.com/RobotLocomotion/pybind11/blame/drake/include/pybind11/cast.h#L1976) which goes to the `cast` you have [here](https://github.com/RobotLocomotion/pybind11/blame/drake/include/pybind11/cast.h#L1968), the `cast` again has `take_ownership` as its policy, similar to my previous comment.  
If we `s/take_ownership/reference` in the aforementioned `cast`, there is no segfault, like we'd expect.
\cc @jamiesnape

EricCousineau-TRI, 2019-05-01:
Ahh... So it's our fork's fault... Oops!

Perhaps it would be good to update the SFINAE for the `T&&` cast to add `!is_pointer<T>` - that way pointers fall back to the correct casting mechanism.
Can I ask if you can briefly try that out?

EricCousineau-TRI, 2019-05-16:
Per f2f, I will try deleting the `py_reference` for `T* f()` from the PR.

PR #11409 (2019-06-20): Remove py_reference from a few casts
Closes #9398

Requires:
- [x] https://github.com/RobotLocomotion/pybind11/pull/28

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11409)
closed 2019-06-20 by m-chaturvedi