RobotLocomotion/drake #9404: Add symbolic formula for isinf / isfinite

jwnimmer-tri, 2018-09-07:
In general, we should support most common functions from https://en.cppreference.com/w/cpp/numeric/math in our Expression overloads.  Mostly we add only when needed, beyond a nice starter pack that we already have on master.

Recently I needed `isinf` in work that ended up not making it to master.  I should add `isinf` anyway, since its could easily be needed in Drake for, e.g., simulation.

jwnimmer-tri, 2018-09-12:
\CC @soonho-tri FYI

soonho-tri, 2018-09-12:
@jwnimmer-tri, please feel free to assign this to me if you haven't started implementing the feature.

jwnimmer-tri, 2018-09-12:
Sure, thanks.

While we could make this its own formula term, I just thinking of just reusing Eq / Lt / Gt formulas comparing the argument with `ExpressionConstant{numeric_limits<double>::infinity()}`, or -inf, etc.

soonho-tri, 2018-09-12:
I like the idea. I'll implement that.

PR #9463 (2018-09-19): Add symbolic isinf and isfinite
Close #9404

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9463)
closed 2018-09-19 by jwnimmer-tri